nanoframework/Home #103: Wire protocol proof-of-concept

nfbot, 2017-08-30:
From nf-interpreter created by [cw2](https://github.com/cw2) : nanoframework/nf-interpreter#8  

Implement wire protocol proof-of-concept library

- [x] Part of nanoBooter PoC (#7)
- [x] Based on NETMF implementation
- [x] No endianness handling (will be done at PC side)
- [ ] Reduce size of header field (-> 4 bytes),
- [ ] CRC optional (needed only for unreliable transports; kudos @Becafuel :clap: )
- [ ] Smaller/configurable payload buffer - now it uses 2048 bytes
- [ ] review flags (required?)
- [ ] review sequence fields (required? with the current format?)

networkfusion, 2018-02-04:
Where do we think we are with the... I would guess the POC is complete, but should it be closed or split out?

josesimoes, 2018-02-05:
That question is tricky... Pherhaps we should manage all this with projects and convert the task into issues when they are in the works....

josesimoes, 2018-02-05:
Move partial tasks to Wire Protocol project.

closed 2018-02-05 by josesimoes