nanoframework/Home #105: nano Portable Executable (nPE) file format

nfbot, 2017-08-30:
From nf-interpreter created by [cw2](https://github.com/cw2) : nanoframework/nf-interpreter#40  

Based on NETMF's [PE](https://github.com/NETMF/netmf-interpreter/wiki/PeFileFormat), but without its obsolete/unused features. Planned modifications:
* Follows ECMA-335 specification for metadata tables,
    * Extending metadata tables is needed to support generics
* No padding information, always aligned to 32-bit boundary (even on the disk),
* No endianness information - handled on the PC side (?)

https://github.com/dotnet/corefx/tree/master/src/System.Reflection.Metadata/src/System/Reflection/Metadata/Ecma335

- [ ] Does Roslyn provide extensibility point for a custom assembly emitter?
    - [ ] If yes, implement Roslyn extension to produce nPE files
    - [ ] If not, implement managed tool to transform .NET dll into nPE (with MSBuild task wrapper?)
- [ ] Update assembly loader C++ structures and functionality
- [ ] Unit tests

josesimoes, 2019-08-12:
Still active

josesimoes, 2019-10-11:
still active

StephenCardinale, 2019-10-11:
Jose,

 

I’m still active. I haven’t had much time lately but plan on getting back to nF soon. The hard part will be to convince Christophe to become enthused again.

Ellerbach, 2023-04-18:
@josesimoes and @StephenCardinale looking at old issues to clean a bit things. Is this one still active?
It's 3.5 years old ;-)

josesimoes, 2023-04-19:
Yes. Most of this will be cleared with the ongoing work so hopefully we'll be able to close this soon.