nanoframework/Home #106: [Draft] Roadmap

nfbot, 2017-08-30:
From nf-interpreter created by [cw2](https://github.com/cw2) : nanoframework/nf-interpreter#55  

# Major goals for the release
* Easy to get up and running on new hardware by leveraging existing HAL and platform code bases
    * ChibiOS/HAL and CMSIS-RTOS compatible ChibiOS/RT
* Easy to build
    * CMake, [built-in support in Visual Studio 2017](https://blogs.msdn.microsoft.com/vcblog/2016/10/05/cmake-support-in-visual-studio/)
* Easy configuration

# High-level engineering plan
* Architecture neutral hardware agnostic libraries for core components
    * Bootloader (nanoBooter)
    * Core Language Runtime (nanoCLR)
    * Framework Class Library (nanoFCL)

## Runtime
* Standalone
* Can be run from/as RTOS task
* Minimal hardware requirements (memory, peripherals)
* Parametrized (e.g. flash memory layout passed via a parameter/callback)
* Configurable at compile time
* Depends only on corlib
* No endianness handling (in a rare case when it's needed it is done on the PC side)

## Framework
* nanoFCL customizable
    * Improved object model to support modern hardware peripherals
* Visual Studio 2017 integration
* Visual Studio 2015 integration  ❓
* Visual Studio Code integration ( ❓ this would be for native development or for C# too ?)
* (?) USB CDC default transport

---
## Next
* Support for additional CMSIS-RTOS compatible systems,
* Improve and optimize wire protocol
   * CRC optional - not required for reliable transports (USB, TCP/IP), needed only for serial (?)
   * Configurable payload buffer (to reduce memory footprint) 
   * Remove unused flags and sequence fields (?) ❗️ (sequence fields are important/relevant for operations that require more than one message+reply, one example is flashing a device)

## Future
* Generics
    * PE format enhancements
* .NET Standard library compliant (strict subset)
* Support for native RTOS tasks that run in parallel with nanoCLR
* Deterministic GC
* Concurrent interpreter
* Partial JIT or AOT

c-born, 2019-05-16:
Is this in need of a review and update?
At least to add in Visual Studio 2019 integration...

josesimoes, 2019-05-16:
Actually most of the goals here are already meet, except the ones under "future".

I'm closing it as it doesn't make much sense to remain open. A roadmap has to be available elsewhere.

closed 2019-05-16 by josesimoes

c-born, 2019-05-16:
Good idea, I'd like to see a roadmap somewhere, including what has been achieved and when.
Good for moral to be able to see your progress!