nanoframework/Home #107: Add CMake build for Visual Studio 2017

nfbot, 2017-08-30:
From nf-interpreter created by [cw2](https://github.com/cw2) : nanoframework/nf-interpreter#87  

Visual Studio 2017 has [built-in support for CMake projects](https://blogs.msdn.microsoft.com/vcblog/2016/10/05/cmake-support-in-visual-studio/)

### Use case

- A developer clones nF git repo,
- opens the root folder in Visual Studio 2017,
- selects a board/target configuration,
- performs build.

networkfusion, 2018-02-01:
this is now possible: https://blogs.msdn.microsoft.com/vcblog/2017/10/23/arm-gcc-cross-compilation-in-visual-studio/

however it is probably worth waiting for the official release.

networkfusion, 2018-03-05:
VS15.6 Officially released...

networkfusion, 2018-03-07:
[image]

15.6 doesnt seem to even like opening the folder, it spawns an exponential amount of processes until VS starts throwing memory errors...

josesimoes, 2019-02-06:
This has been added to nf-interpreter in nanoframework/nf-interpreter#1151.
Despite being considered experimental and still requiring some manual tweaks, which is not optimal, this is working.

closed 2019-02-06 by josesimoes