nanoframework/Home #109: ESP32 Support

nfbot, 2017-08-30:
From nf-interpreter created by [cw2](https://github.com/cw2) : nanoframework/nf-interpreter#136  

I am happy to announce that ESP32 support is now under active development.

The implementation is based on the official [Espressif IoT Development Framework](https://github.com/espressif/esp-idf) (ESP-IDF), initially for a generic [ESP-WROOM-32](https://espressif.com/en/producttype/esp-wroom-32) module.

The following is a rough roadmap:

- [x] Initial experiments (toolchain setup, build environment, deployment, debugging, diagnostics)
- [ ] nanoBooter port
  - [ ] :question: implement as a part of 2nd stage bootloader or a separate [3rd stage] ?
- [ ] :running_man: nanoCLR port

> Note: The nanoCLR is being done now (before bootloader) to synchronize with ongoing ChibiOS HAL work.

ESP-IDF has many features that make it very good fit for nanoFramework (the list is not complete, and in no particular order of importance):
- GCC-based toolchain,
- Highly componentized and customizable build system,
- Bootloader is a separate application,
- Customizable flash memory partition table,
- Over-the-air updates

AdrianSoundy, 2017-12-25:
I would like to announce that ESP32 support is now really on its way to being developed and the initial running version is now complete with support for  :- 

* GPIO and events ( Tested )
* SPI ( still to be tested with Wrover display )
* I2c (Tested with Oled display)

The initial version will not have any networking support, which is really the whole point of the ESP32.

This version will be just the NanoClr using the ESP32 IDF bootloader instead of Nanobooter.
The deployed managed code is currently written to a separate flash partition.

piwi1263, 2017-12-25:
@AdrianSoundy this is wonderful news to hear on a 1st X-mas day.

josesimoes, 2017-12-26:
@AdrianSoundy great news indeed! 👏 💯  
Let's have that PR to get the code in the repo so others can help. 😉

networkfusion, 2018-02-01:
@josesimoes since the code is now in the develop branch, is this not complete?

josesimoes, 2018-02-01:
The support for it is there. Despite there is still a lot of work ahead to reach the full potential of the chip the support for it is there. So let's close this and kudos (again) to @AdrianSoundy for it!

closed 2018-02-01 by josesimoes