nanoframework/Home #112: Rework HAL time functions to achieve compatibility with both .NET ticks and CMSIS SysTicks

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#159  

In .NET world a point in time is represented with a [DateTimeOffset Structure](https://msdn.microsoft.com/library/system.datetimeoffset) which has a time component measured in 100-nanosecond units.

The native part of nanoFramework is based on CMSIS whose time base is provided by [SysTick](http://arm-software.github.io/CMSIS_5/RTOS/html/group__CMSIS__RTOS__KernelCtrl.html#gad0262e4688e95d1e9038afd9bcc16001) that is platform and implementation dependent.

For embedded systems it doesn't make sense to waste resources with timekeeping and time management in nanoseconds (requires using timers larger than 32 bits, using 64 bits vars, use 64 bits math, etc).

The HAL functions have to be revisited and reworked to achieve compatibility between the two worlds. At the same time it should be made perfectly clear when a .NET _tick_ is being used/required and when a CMSIS _tick_ is being used/required. This will require renaming of function/arguments/vars.

HAL functions that are "internal" to the code should use the CMSIS _ticks_ (to be resource wise).

HAL functions that are used by .NET API should use the .NET _ticks_. If a conversion between _ticks_ is required (at platform level) the CMSIS functions are to be used in order to find out what is the frequency and tick real time value for correct calculations.

Ellerbach, 2023-04-18:
@josesimoes this has been done as far as I remember. If yes, please close it, so we can get things clean. Thanks!

josesimoes, 2023-04-19:
Actually no... this is still work to be done....

Ellerbach, 2026-09-28:
@josesimoes isn't it what you've been implementing recently with the Pico addition?

josesimoes, 2026-09-28:
Still no... That was about bringigmng the Pico RTC into ChibiOS.