nanoframework/Home #118: Exception with "N0" format string and integer value of 0

nfbot, 2017-08-30:
From nf-interpreter created by [MikroBusNet](https://github.com/MikroBusNet) : nanoframework/nf-interpreter#285  

The following code will throw a 'System.ArgumentOutOfRangeException' in mscorlib.dll :

```csharp
private static uint myVar = 0;

public static void Main()
{
    Debug.Print("myVar = " + myVar.ToString("N0"));
}
```
If myVar is 0 and precision is 0 for the 'N' format string, then the FormatNative() method is returning an empty string, which will raise the exception in the InsertGroupSeparators() method (in Number.cs), since original.Length is 0 :
```csharp
int digitsStartPos = (original[0] == '-') ? 1 : 0;
```
We can see the same behaviour for all integral types except long and ulong.

MikroBusNet, 2017-12-19:
Issue is still present, as the quick&dirty fix has been removed.

MikroBusNet, 2017-12-19:
Looks like it is the expected behaviour, in fact. See last sentence :

`For integer specifiers (d, i, o, u, x, X): precision specifies the minimum number of digits to be written. If the value to be written is shorter than this number, the result is padded with leading zeros. The value is not truncated even if the result is longer. A precision of 0 means that no character is written for the value 0.`

Taken from : http://www.cplusplus.com/reference/cstdio/printf/

josesimoes, 2017-12-20:
Waiting for the PR to be submitted on the CoreLibrary repo.

closed 2017-12-21 by josesimoes