nanoframework/Home #120: Add support for Bitmaps

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#304

josesimoes, 2018-09-11:
VS extension now has the capability to process managed resources. Bitmap resources are still disabled because of lack of support from the framework. nanoframework/nf-Visual-Studio-extension#289

josesimoes, 2018-10-18:
Image resources can now be added to a resource collection.

josesimoes, 2019-02-06:
Work on this has started in nf-interpreter develop-ui branch.

josesimoes, 2019-08-26:
still active

TerryFogg, 2019-10-18:
Bitmap can be supported with the new graphics code but the extension does not create the correct resource class GetBitmap(..) and does not pass through the correct resource kind value of 2 to the underlying corlib_native_System_Resources_ResourceManager.cpp

josesimoes, 2019-12-17:
WIP

AdrianSoundy, 2020-05-25:
Wip

closed 2020-05-26 by AdrianSoundy