nanoframework/Home #121: Add support for fonts

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#305

josesimoes, 2018-09-11:
VS extension now has the capability to process managed resources. Font resources are still disabled because of lack of support from the framework. nanoframework/nf-Visual-Studio-extension#289

josesimoes, 2019-02-06:
Work on this has started in nf-interpreter develop-ui branch.

josesimoes, 2019-03-27:
Active work on this feature at #468.

TerryFogg, 2019-10-18:
Font can be supported with the new graphics code the extension does not creates the correct resource class GetFont(..) but does needs editing to correct the namespace from Microsoft.Spot to nanoFramework.UI, it does pass through the correct resource kind value of 2 to the underlying corlib_native_System_Resources_ResourceManager.cpp

AdrianSoundy, 2020-05-25:
Wip

closed 2020-05-26 by AdrianSoundy