nanoframework/Home #127: Cleanup CLR_RT_WellKnownTypes in TypeSystem.cpp

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#320  

There are several members on that struct that don't make sense anymore, need to be renamed and/or are taking space unnecessarily.
Microsoft.SPOT needs renaming
Microsoft.SPOT.Touch doesn't exist
Microsoft.SPOT.Cryptoki doesn't exist
etc

MikroBusNet, 2017-09-15:
Could the following be completely removed or should the texts be replace with NULL ?
`TIL( "Microsoft.SPOT.Hardware" , "I2CDevice"                     , m_I2CDevice                                          ),
    TIL( NULL                      , "I2CReadTransaction"            , m_I2CDevice__I2CReadTransaction                      ),
    TIL( NULL                      , "I2CWriteTransaction"           , m_I2CDevice__I2CWriteTransaction                     ),`

josesimoes, 2017-09-22:
@MikroBusNet that's on the type lookup index.... a bit more tricky... we sure could start by removing what we don't have and correct what is different.

closed 2017-12-14 by josesimoes