nanoframework/Home #129: Implement native code helper to generate a new Guid

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#323  

From nanoframework/nf-class-libraries#40

This is `GenerateNewGuid___STATIC__SZARRAY_U1`.

The method should return an array with 16 random byte, ready to generate a valid Guid.

closed 2018-07-25 by josesimoes

networkfusion, 2018-07-26:
Is this also done for stm32 devices? I saw no mention of it in the PR?

josesimoes, 2018-07-26:
> Is this also done for stm32 devices? I saw no mention of it in the PR?

Yes it is. 

Granted, the mention to ESP32 in the PR comment is probably misleading... 😅  What was added in the PR was the code to generate the GUID _AND_ the random generator implementation for ESP32 (which was missing until now). 😜