nanoframework/Home #130: Update Travis yaml to have targets build with different API configurations

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#325  

Ideally the build image should include all the available APIs to properly test for issues at build stage.
This is not possible because (probably!) only targets with larger flash can hold all the APIs.
The best approach seems to be duplicate the existing targets but changing the API combination that should be included in the image. 

E.g. have a NUCLEO901 include the Windows.Devices.Gpio and Windows.Devices.SPI on one image, add a new configuration to include Windows.Devices.SPI and Windows.Devices.I2C on another, etc.

networkfusion, 2018-02-01:
@josesimoes is this now complete or is there still a better way of achieving it?

josesimoes, 2018-02-01:
Yes and no... The initial idea was to create build jobs with a different mixes of build options. This is not being done.  
All the available ones have a fix set of options.
This would be for a daily or weekly build that could test several combinations and is necessarily taking a looing time to complete execution.

josesimoes, 2019-01-07:
This is not required anymore. 
Most of the targets have all features enabled anyways so this should be proof of _buildability_ enough.

closed 2019-01-07 by josesimoes