nanoframework/Home #136: Work on Monitor_FlashSectorMap

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#382  

nanoBooter is not being able to report the Flash map because the code in the command is commented in WireProtocol_MonitorCommands.c.

Uncomment and follow the implementation of CLR_DBG_Debugger::Monitor_FlashSectorMap to make it work

networkfusion, 2018-02-04:
Also BlockStorageDevice requires ```#include <nanoPAL_BlockStorage.h>```

still fails at ```const DeviceBlockInfo* deviceInfo = device->GetDeviceInfo(device);``` though which has lots of FixMe comments in block storage...

MikroBusNet, 2018-02-04:
I think that it is a remnant of NETMF code and that the code indeed needs some tweaking in this section to be able to work with nf.

closed 2018-03-16 by josesimoes