nanoframework/Home #141: Port and build for STM32F411RE

nfbot, 2017-08-30:
From nf-interpreter created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-interpreter#413  

@valoni as I didn't want to hijack the .NETMF issue, so I'm opening this here were it belongs.

Regarding your questions on NETMF/netmf-interpreter#538:

- indeed there is a community board for the STM32F411RET6, @piwi1263 is the author
- you can fin the build instructions along with other documentation in the docs folder [here](https://github.com/nanoframework/nf-interpreter/tree/master/docs)
(you'll find that you don't nee much 😉 )
- as the documentation is a bit scarce you'll probably need some guidance suggest you join our Slack community by filling in the form [here](https://nanoframework.wordpress.com/slack-invite-form)
- you'll the VS 2017 extension too that you can install directly in VS by searching "nanoFramework" in the Tools and Extensions Window.

Trust this can, at least, get you started. Anyways you'll get all the help that you ask for here on in the Slack community.

josesimoes, 2017-09-05:
@valoni I believe you've managed to successfully build this target. 
If yes, then this can be closed. Please confirm.

valoni, 2017-09-05:
YES after some tips and info i received i learn more to manage sources
- latest build i build without problem
- also i can manage problem with overleap size
- activated SPI=ON without problem on DEBUG mode for both boards
(Discovery STM32F407VGT6 and NUCLEO STM32F411RET6)

but the problem i face is

at home i have pc with VS2017 (and i'm there just later or when i have
available time)

at work when i am and on faculty when i tech students we have VS2015
(there i have time but i could not change development tools so i cant use
it for development and test since need VS2017 except VS2017 code)

so for confirmation i managed to build today all those platforms for Nucleo
STM32F411RET6

*1.) nanoFramwork 0.2.0 *

*2.) NET Microframwork 4.4.0.0*(today i build) #
 i dont know why it is not added by default on Discovery is OK but on
Nucleo NO so need to check for
   -  Microsoft.SPOT.Hardware.PWM,4.4.0.0
   -  Microsoft.SPOT.Hardware.OneWire,4.4.0.0

*3.) TinyCLR OS 0.6.0* (it can use now ST-LINK programmer- as SERIAL
Debug/Deploy with TinyCLR no need more for PA11/PA12 connector)

valoni, 2017-09-05:
Please close this issue :)
--
Sent from myMail for Android Tuesday,

josesimoes, 2017-09-05:
Thanks for letting us know about your successes. 
Suggest that you ask for help on that other issue that you're facing now on the appropriate Slack channel.

closed 2017-09-05 by josesimoes