nanoframework/Home #173: There should be an option to erase deployment region

nfbot, 2017-08-30:
From nf-Visual-Studio-extension created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-Visual-Studio-extension#135  

Sometimes when deploying a "bad" managed application assembly can get the device locked into an unusable state.
An option should exist in Device Explorer to erase the deployment area. This should be available when the device is present in both nanoCLR and nanoBooter modes.

MikroBusNet, 2017-09-15:
In nanoCLR mode, the bad application would be already launched, thus preventing its deletion from the device manager, to me.
Or did I miss something ?

josesimoes, 2017-09-17:
That is correct. 
It's to help the developer in situations where a bad is hanging the board.
If the button to remain in booter is pressed this allows the deployed app to be erased, the CLR then can load fine again and then a new app can be deployed.
Otherwise these situations end up requiring a flash erase and redeploy booter and CLR.

MikroBusNet, 2017-09-17:
Then your first comment should only talk about "nanoBooter mode" ;)
I was wondering how it could work in nanoCLR mode because it was too late in the process.

closed 2018-01-03 by josesimoes