nanoframework/Home #196: Readme on I2M_OXYGEN_NF seems to be wrong

nfbot, 2017-08-30:
From nf-Community-Targets created by [josesimoes](https://github.com/josesimoes) : nanoframework/nf-Community-Targets#6

networkfusion, 2017-10-19:
@piwi1263  any update on this? I don't mind sanity checking for you but don't have the knowledge of the board to write the document...

piwi1263, 2017-10-19:
This is WIP, started some time ago, but got to a hold due to the very dynamic environment it has to sync to.

networkfusion, 2017-10-19:
This issue applies more to the ReadMe more than a working build... I have created a new issue for that (and all other community targets). It would be good if we could get this one closed though as if the ReadMe is correct it will (maybe) help others change the rest...

josesimoes, 2018-01-29:
@piwi1263 I'm pretty sure this is correct now... 😉  can this be closed?

closed 2018-01-30 by piwi1263