nanoframework/Home #221: nanoCLR not booting on STM32F769I

networkfusion, 2017-09-04:
## Details about Problem

nanoCLR does not boot (cannot debug/ USB does not show in device explorer) on the STM32F769I. This has been traced to commit [3d07ab2] #459 as it works up until this commit.

VS version (if appropriate):

OS version (i.e. win10 v1607 (14393.321)):

Worked before? If so, with which nanoFrameowrk version:

## Detailed repro steps so we can see the same problem

1. Build and run latest versions of nanoBooter and nanoCLR on the STM32F769I

2.

...

## Other suggested things

### Sample Project (if applicable)

Very helpful if you can zip a project and paste into this issue!

### Screenshot (if applicable/relevant)

Very helpful if you send along a few screenshots to help visualize the issue!

MikroBusNet, 2017-09-04:
This commit is related to the Convert namespace in Corlib. The code in this assembly is not used from the native portion of the firmware !?

josesimoes, 2017-09-04:
Found the issue: it's in the call to strtod brought with the inclusion of `NativeToDouble___STATIC__R8__STRING`.

Steps to "reproduce":
1. comment line 35 of corlib_native_System_Convert.cpp `stack.SetResult_R8 (strtod(_string, nullptr));`
2. rebuild and run, STM32F769I target loads as expected.

Further investigation:  adding strtod increases flash size by 56.096 bytes.
It seems too much... Worth considering to revert this back to the previous native implementation.

MikroBusNet, 2017-09-04:
How can this code prevent loading the CLR ?

josesimoes, 2017-09-04:
It's causing an hard fault on the ARM core. Haven't dig deeper.

MikroBusNet, 2017-09-04:
What I mean is : how can non-called code cause a fault ?

networkfusion, 2017-09-04:
Perhaps: http://www.microchip.com/forums/m922599.aspx#922599

change 

`double strtod (const char *, char **);`

to

`long double strtod (const char *, char **);`

MikroBusNet, 2017-09-04:
That would require changing the signature of the method and use *strtold()*, I think.
Trying without the FPU option should be easier, however.

I do not have this board so I cannot test.

josesimoes, 2017-09-05:
Digging deeper into the boot revelas that the root cause is a bad region definition. This is calculated in linker file and shows as `ram_areas`.
The inclusion of strtod _blows up_ the counters on the linker file. It could be with any other call. The thing is that it overflows a certain threshold of code going into FLASH.

josesimoes, 2017-09-05:
The root cause for this is an incorrect declaration of several regions on linker file which have their names and alignment incorrect. This causes the `ram-areas` to be miss calculated.
For some reason (possibly because the constructors weren't really used?) this hasn't surfaced before. 
The addition of `strtod` (which is a complex function and includes several calls to inner functions) has triggered this.

closed 2017-09-05 by josesimoes