nanoframework/Home #222: Evaluate strtod against native implementation

josesimoes, 2017-09-05:
`strtod` is being included by a call in `NativeToDouble___STATIC__R8__STRING`.
This is 
It turns out that the inclusion of this (despite being the newlib nano implementation) increases FLASH size by 7.504 bytes and and RAM by 60 bytes.

Need to evaluate if we should revert this to the native implementation or leave it as it is and possibly optimize it when we move forward with the _flavoured_ mscorlib.

Pros for having a native implementation:
- fastest execution

Cons:
- increased size in FLASH and RAM
- unless there is an extensive use of calls to conversion methods, the poorer execution time shouldn't be really an issue

MikroBusNet, 2017-09-05:
Do not beat around the bush and put back the managed code.

josesimoes, 2017-09-05:
Genuinely asking for folks commenting on this.
If this was black and white I wouldn't be asking.

For example if you are working with a target that has plenty of flash there is no big deal on using the native crt implementation. Now, if you are squeezing nF on a STM32F091 that's a completely different story.

MikroBusNet, 2017-09-05:
If the managed code size is smaller than the native one, then there is no question.

Btw, if we are talking about very small boards, then we can gain an aditionnal 1.200 bytes by removing the _strtoll()_ method from the native code as well (in the same assembly).

josesimoes, 2018-06-21:
We can easily provide the best of both worlds!
A CMake option exposing this allows the developer to choose which one best fits the platform he/she's targeting and the requirements of the application.
No change is required at mscorlib in what concerns the native calls.

### CMake option: SUPPORT_ANY_BASE_CONVERSION

- ON: all methods in Convert() class support converting from any base
- OFF: all methods in Convert() class support converting from base 10 and 16 (most convenient in embedded systems)

### Image size comparison
This is for the STM32F769I_DISCOVERY target complied in debug flavour.

- ON (any base): 210.590 bytes
- OFF (base 10 and 16): 204.954 bytes (less 5.636 bytes 👍 )

closed 2018-06-21 by josesimoes