nanoframework/Home #241: STM32F769I (and possibly others) not showing deployed assemblies

networkfusion, 2017-09-27:
## Details about Problem
Possible regression... Deployed assemblies are not shown for (at least) the STM32F769I when clicking "Device Capabilities"

VS version (if appropriate):
15.3.5
OS version (i.e. win10 v1607 (14393.321)):
Windows 10
Worked before? If so, with which nanoFrameowrk version:
Yes, but not sure how recently...

## Detailed repro steps so we can see the same problem

1. Deploy nanoBooter and nanoCLR through VSCode (F5/debug)

2. Connect and deploy assemblies onto the device using VS2017

3. Restart device a few times and make sure program is running

4. Connect and click Device Capabilities

...

## Other suggested things

### Sample Project (if applicable)

Very helpful if you can zip a project and paste into this issue!

### Screenshot (if applicable/relevant)

Very helpful if you send along a few screenshots to help visualize the issue!

MikroBusNet, 2017-10-01:
Same for Quail.
I don't know for other boards.

josesimoes, 2017-10-04:
### Test # 1 
1. Mass erase device with ST Link 
2. Load nanoBooter with VS Code (confirmed LED blink)
3. Load nanoCLR with VS Code (confirmed running with SWO viewer)
4. Open Blinky from nf/samples
5. Connect target in Device Explorer (VS Extension v0.1.63 installed from marketplace after uninstall and reboot)
[image]
6. Deploy Blinky 
[image]
7. Device capabilities don't change, LED doesn't start to blink.
8. RESET button pressed and device reconnected in Device Explorer 
9. Assemblies are listed and managed app runs (LED does blink).
[image]

### Test # 2
1. Mass erase device with ST Link 
2. Load nanoBooter with VS Code (confirmed LED blink)
3. Load nanoCLR with VS Code (confirmed running with SWO viewer)
4. Open DebugGC.Test from nf/samples
5. Connect target in Device Explorer (VS Extension v0.1.63 installed from marketplace after uninstall and reboot)
6. Deploy DebugGC.Test
[image]
7. Device capabilities don't change, LED doesn't start to blink, no output to SWO viewer.
8. RESET button pressed 
9. LED does blink and GC output keeps showing in SWO viewer
10. Hit connect button in Device Explorer 
11. VS hangs with button pressed
[image]
12. Need to hit RESET button on target to have VS back
13. Further attempts to reconnect produce the same result
14. Need to mass erase flash to be able to reconnect to device again

## Conclusions and discussions

a. The behaviour on 7. is expected because there was no app loaded on the target device therefore the execution engine was exited and can't be restarted unless a hard reset occurs. Possible solution would be to implement a conditional reset command on the deploy provider: if there _was_ no assembly deployed issue a hard reset command, if there were assemblies deployed issue a reset for CLR only command. 

b. It seems that the forced execution of the GC prevents the smooth operation of nanoCLR. Follows investigation about this hypothesis.

networkfusion, 2017-10-05:
Can you also test this with SWO off... I will also try it again with the details you have provided... I have also noticed the GC bug...

For reference, I will edit this comment with the config used when I am at my Dev PC...

josesimoes, 2017-10-05:
Thanks for helping out with this! 👍

I've commented the printf there so I'm reasonably sure that it's not because the SWO.

On the GC, I've already pinpoint this to HeapCompactation(). Now going to dig deeper in there.

Please try to replicate the test # 1 above at your end to see if you get the same results.
(BTW, I've added the Blinky sample to the samples repo)

Apart from the different behaviour on the very first programming (because it's not actually executing anything) I'm suspecting that this can be corrected by implementing those different reboot option ls after deployment.

networkfusion, 2017-10-05:
TEST 1: 
When using "Debug" for nanoCLR, the assemblies are shown correctly.
When using "MinSizeRel" for nanoCLR the issue is present, i.e. assemblies are NOT shown (but program is running)
I have not tried "RelWithDebInfo" or "Release".

josesimoes, 2017-10-05:
Interesting... If the program is running it means that everything was deployed properly.
Because it doesn't have the DEBUG definitions a lot of stuff is removed. Maybe this is the behavior by design... After all when you are not debugging no one will ever see that anyways...
A lot of stuff is there only to help visualize things and give insight to the developer on what is going on.

networkfusion, 2017-10-05:
Perhaps this is a documentation issue then... But I cant help feel that at release, Assemblies and their version are important for establishing whether the firmware/program should be upgraded whether manually by the "user" or remotely...

@MikroBusNet which release type were you using????

josesimoes, 2017-10-05:
> But I cant help feel that at release, Assemblies and their version are important 

Absolutely! And that always exists. Otherwise it wouldn't be able to do basic stuff such as resolving types and such, which is obviously doing correctly. 
It's probably a matter of how that particular command gets executed or how it's output to the debugger... Don't know for sure but that seems to be related with the build flavor.

MikroBusNet, 2017-10-06:
>@MikroBusNet which release type were you using????

MinSizeRel + RTM

closed 2017-10-17 by josesimoes

josesimoes, 2017-10-17:
@networkfusion after all there is no bug with `Heap_Compact()`. The real issue was with the wrong size of the managed heap. 
As it tried to access an invalid memory address an hard fault was thrown.
This only occurred after the move of the managed heap region to the end of the RAM. Before that the managed heap region was placed right before the CRT heap and  `Heap_Compact()` was (wrongly but happily) touching the first address of the CRT heap. This didn't cause any hard fault because it was valid address but was messing up with whatever var was located at that address.

The fix is in nanoframework/nf-interpreter#534.