nanoframework/Home #269: Unit testing

networkfusion, 2017-12-29:
## Details about Problem
Unit testing is important in this day and age, yet we currently rely on build completion (appveyor) and manual testing for most of our code. This is particularly hard in embedded development and complicated by using an interpreter but we must find a way of implementing good practice in this area... Some checkpoints are:
[] Create a CI/CD pipeline for all projects (git submodules may help here?!)
[] Find a good unit testing suite for nf-interpreter (c/c++ code)
[] Find a good platform for testing (VSTS?!)

josesimoes, 2020-04-29:
pinging @SandorDobos

PlasmaEye, 2020-10-10:
I've started to look at this and wanted to start with the [System.Device.I2c](https://github.com/nanoframework/lib-System.Device.I2c) library. I was hoping to add an issue to that repo so that I could separate any discussions surrounding its testing from this repo. Would it be possible to have the System.Device.I2c's settings changed so that issues could be created for it?

josesimoes, 2020-10-12:
> I've started to look at this and wanted to start with the [System.Device.I2c](https://github.com/nanoframework/lib-System.Device.I2c) library. 
That's great that you're looking into this.
The idea behind the Unit Test is to have a test framework based on https://github.com/microsoft/vstest along with integration test running on real hardware.
So this should be general and not specific to I2C or "just" this or that component.

I was hoping to add an issue to that repo so that I could separate any discussions surrounding its testing from this repo. Would it be possible to have the System.Device.I2c's settings changed so that issues could be created for it?
>> The reason for having only on place for issues is precisely to concentrate them. The project is rather complex and is split on multiple components. We would risk people opening issues in the wrong repo and also, frequently one issue touches multiple components.
Also we rather have discussion in Discord instead of using github issues for this. It's cleaner and much more efficient and productive. 
If you're not already there, may I invite you to join our Discord community [here](https://discordapp.com/invite/gCyBu8T)? 😃 
(when there head to #unit-functional-test channel)

PlasmaEye, 2020-10-13:
That makes sense. Thank you for clarifying.

closed 2021-02-25 by josesimoes