nanoframework/Home #270: VS2017 hangs after deployment

MikroBusNet, 2018-01-01:
## Details about Problem
After successfull deployment and board reboot, VS2017 UI becomes unresponsive after the message "nanoDevice is initialized".
VS has to be killed manually.

VS extension is 3.8.12 and Visual Studio is 15.4.3.

This was tested on a Quail and on F429i-Disco and is happening after two or three deployments.

The managed program is loaded and running, however. Only VS UI is hung.

### Screenshot
[image]

After this, the board reboots and runs the managed program. But VS2017 is unresponsive.

Stabilitronas, 2018-01-03:
I have VS 15.5.2, but get pretty much the same result. It went to a point when I can't even deploy to the Quail anymore.

josesimoes, 2018-01-04:
I'm seeing this too when the board is not working or is irresponsive.
Adding to this (and this is only sometimes) I have a hard time terminating VS with the Task Manager. Have to wait a lot of time and sometimes even insist on terminating the task.

closed 2018-02-05 by josesimoes