nanoframework/Home #274: PWM does not work with latest builds from repo

MikroBusNet, 2018-01-07:
## Details about Problem
If you download firmware from the repo's readme page (automated builds), then pwm does not work and throws a _NotSupportedException_ on every pwm method.

Firmware build locally after downloading source from the repo is working fine, though.

josesimoes, 2018-01-07:
Easy peasy!
I've just checked appveyor yaml and, guess what, PWM isn't on on BUILD_OPTIONS for any board. 
No wonder that it gives you unsupported exception. 😜

josesimoes, 2018-01-08:
Fixed with [1ecb2f6](https://github.com/nanoframework/nf-interpreter/commit/1ecb2f64b4ef1b62dfe4c2da1c1f16f371050fbe) and [7210659](https://github.com/nanoframework/nf-interpreter/commit/721065918c05112e993e47a9453fd50b922d9101).

closed 2018-01-08 by josesimoes