nanoframework/Home #275: emit error on compile

AdrianSoundy, 2018-01-08:
## Details about Problem

When trying to compile a foreach loop with a string the compiler gives the error:-

CSC : error CS7038: Failed to emit module

nanoFramework area (C# code | Visual Studio extension )

## Detailed repro steps so we can see the same problem

1.  Try to compile following sequence
      
   string test = "abc";
   foreach(char c in test)
   {
   }

piwi1263, 2018-01-08:
Remember this is UWP not NetMf, you can't do it that way. I've run into that too while porting the SSD1306 driver. Try line 307 before doing the foreach loop.

[image]

AdrianSoundy, 2018-01-09:
In a desktop UWP app it works.

I found the problem in the end and used a normal for loop which worked

I'll change it to your way as it looks a cleaner solution.

Adrian

josesimoes, 2018-01-09:
The root cause for this should be an issue with the string enumerator.
In the mean time the workaround suggested by @piwi1263 seems to be the best approach.

josesimoes, 2019-04-01:
This is happening because `String` doesn't implement `IEnumerable`.
Adding support for this would increase the deployed assembly size.
Because an workaround for this exists, we are better keeping it as it is.

This may change in the future if we decide to add `IEnumerable` implementation to `String`.

closed 2019-04-01 by josesimoes