nanoframework/Home #277: Visual Studio extension deploy problem

AdrianSoundy, 2018-01-18:
## Details about Problem

If you have a nanoFramework Solution with a class library project and a Application project.

If you try to deploy the application it will first deploy the class library project and all related assemblies and then deploy the Application project and all related assemblies including class library.

nanoFramework area ( Visual Studio extension )

VS version (if appropriate): 15.5.4
OS version (i.e. win10 v1607 (14393.321)): win10
Visual studio extension : 0.3.8.18

## Detailed repro steps so we can see the same problem

1. Create a solution with a new class library
2. Add a application project to solution, no need to reference class library
3. Deploy application to device.
4. Will deploy both projects
...

## Other suggested things

### Sample Project (if applicable)

Very helpful if you can zip a project and paste into this issue!

### Screenshot (if applicable/relevant)
Build / deploy log from visual studio
[image]

If you need VS solution i will upload it.

AdrianSoundy, 2018-01-18:
I found you can disable that behavior under solution *Configuration manager* by turning off the deploy check for the class library.

The default behavior doesn't make sense for nanoframework as everything has to be deployed in one file.

I suppose this issue can be closed

josesimoes, 2018-01-18:
Yes, this behaviour has been identified before. This is a duplicate of #137.
(you've figured out the solution for it 👏 )
As far as I know, there is no way to have this automated by VS, so the developer has to take care of it.

closed 2018-01-18 by josesimoes