nanoframework/Home #281: VS Extension cannot connect to board after unhandled exception in managed main()

MikroBusNet, 2018-02-02:
## Details about Problem
When there is an unhandled exception in an application's main() method, then the nf extension cannot connect back to the board and requires to reflash the firmware.

If you don't have swo enabled, you can't see or know what happened in your program. So you can't fix the error easily.

josesimoes, 2018-02-04:
I remember seeing this with .NETMF... can someone confirm?
I'm asking to determine if this is something that was inherited with the execution engine or something that was introduced in the reworks.
My understanding of it is that when main exits (which is something that is not supposed to occur), the CLR goes and starts the execution again putting the execution in a tight loop that starves CPU resources leaving the board irresponsive.

josesimoes, 2018-02-14:
This seems to be fixed with VS extension 0.3.29 and nanoCLR 0.1.0-preview.476.

closed 2018-02-14 by josesimoes