nanoframework/Home #282: Gpio events does behave as expected

MikroBusNet, 2018-02-03:
## Details about Problem
Using the sample located in the [Samples repo](https://github.com/nanoframework/Samples) with a F429 Discovery :
- clean FW, debug : looks good (!)
- change blinky loop delay, debug : exception
- clean FW, same modified sample (same modified delay) : looks good (!)
- modify code anywhere, debug : exception

This is one thing. But, more annoying : 
(!) _Looks good_ but it is not really, in fact. The event is thrown **after** the sleep delay in the blinky loop :disappointed:
If you set a delay of 2 seconds, then you will see the green led flash **after** that 2 seconds delay.

josesimoes, 2018-02-05:
Issue reproduced.

josesimoes, 2018-02-09:
Managed to reproduce the exception when adding the event handler.
That only occurs if _only_ if the CLR is reboot. On a hard or power up boot that won't occur. (thanks @MikroBusNet for the tip!).
This is affecting the ChibiOS based targets. The reason for it this is that `EXTConfig extInterruptsConfiguration` is not being reset on CLR reboot, thus keeping the setting from the previous run. On the following runs the code is checking for any handler there (because it can only have one) and as it finds that's taken (from the previous run) it throws an exception.

closed 2018-02-09 by josesimoes