nanoframework/Home #290: Biard hung after GC kicked in

MikroBusNet, 2018-03-03:
## Details about Problem

Once the garbage collector has run, the board becomes unresponsive and, at least for Quail, needs a firmware reflash, as VS extension cannot attach again.

### Sample Project
Using the following code, the GC kicks in at about 380 timer ticks and hangs :
```
using System;
using System.Text;
using System.Threading;
using static System.Console;

namespace TestTimer
{
    public class Program
    {
        private static Timer _timer;
        private static Int32 _timerCount = 0;

        public static void Main()
        {
            try
            {
                WriteLine("Program stated");
                _timer = new Timer(new TimerCallback(HandleTimer), null, 1000, 250);

                for (var i = 0; i < 1000; i++) { var z = 0; Thread.Sleep(1); };
            }
            catch (Exception ex)
            {
                WriteLine("Exception " + ex.Message);
            }
            finally
            {
                WriteLine("Entering infinite loop");
                while (true)
                {
                    Thread.Sleep(100);
                }
            }
        }

        private static void HandleTimer(object status)
        {
            WriteLine("Timer tick : " + (++_timerCount));
            var strb = new StringBuilder();
            strb.AppendLine("Test");
            var str = strb.ToString();
        }
    }
}
```

The timer callback is not doing anything useful but contains some string manipulations so that the GC can run.

> Timer tick : 371
Timer tick : 372
Timer tick : 373
Timer tick : 374
Timer tick : 375
Timer tick : 376
GC: 1msec 6876 bytes used, 160404 bytes available
Type 0F (STRING              ):    108 bytes
Type 11 (CLASS               ):    204 bytes
Type 12 (VALUETYPE           ):    144 bytes
Type 13 (SZARRAY             ):     24 bytes
   Type 07 (I4               ):     24 bytes

networkfusion, 2018-03-03:
Is this the latest dev branch (I.e chibios 18.x.x)?

MikroBusNet, 2018-03-03:
Latest of all except Visual Studio (v15.5.4). FW 552 downloaded from repo.

josesimoes, 2018-03-05:
This is what I see when loading this on a STM32F769I DISCO:

```

nanoCLR (Build 0.0.0.0)

nanoCLR running @ ST_STM32F769I_DISCOVERY built with ChibiOS v18.2.0

Starting...
Created EE.
Started Hardware.
Create TS.
Loading Deployment Assemblies.
Attaching deployed file.
   Assembly: mscorlib (1.0.0.0)   (4304 RAM - 36500 ROM - 21188 METADATA)

   AssemblyRef    =        0 bytes (       0 elements)
   TypeRef        =        0 bytes (       0 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =        0 bytes (       0 elements)
   TypeDef        =     1128 bytes (     141 elements)
   FieldDef       =      264 bytes (     131 elements)
   MethodDef      =     1776 bytes (     888 elements)
   StaticFields   =      108 bytes (       9 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        4 bytes (       1 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =     1653 bytes
   Signatures      =     2544 bytes
   ByteCode        =    13531 bytes

Attaching deployed file.
   Assembly: nanoFramework.Runtime.Native (1.0.0.0)   (304 RAM - 1012 ROM - 615 METADATA)

   As     0 elements)
   MethodRef      =       44 bytes (      11 elements)
   TypeDef        =        8 bytes (       1 elements)
   FieldDef       =        4 bytes (       2 elements)
   MethodDef      =        8 bytes (       4 elements)
   StaticFields   =       24 bytes (       2 elements)

   Attributes      =        0 bytes (       0 elements)
   TypeSpec        =        0 bytes (       0 elements)
   Resources       =        0 bytes (       0 elements)
   Resources Files =        0 bytes (       0 elements)
   Resources Data  =        0 bytes
   Strings         =      150 bytes
   Signatures      =       60 bytes
   ByteCode        =      190 bytes

Resolving.

Total: (3968 RAM - 38316 ROM - 22139 METADATA)

   AssemblyRef    =        8 bytes (       2 elements)
   TypeRef        =       60 bytes (      15 elements)
   FieldRef       =        0 bytes (       0 elements)
   MethodRef      =       84 bytes (      21 elements)
   TypeDef        =     1176 bytes (     147 elementTimer tick : 1
Timer tick : 2
Timer tick : 3
Timer tick : 4
Timer tick : 5
Timer tick : 6
Timer tick : 7
Timer tick : 8
Timer tick : 9
Timer tick : 10
Timer tick : 11
Timer tick : 12
Timer tick : 13
Timer tick : 14
Timer tick : 15
Timer tick : 16
Timer tick : 17
Timer tick : 18
Timer tick : 19
Timer tick : 20
Timer tick : 21
Timer tick : 22
Timer tick : 23
Timer tick : 24
Timer tick : 25
Timer tick : 26
Timer tick : 27
Timer tick : 28
Timer tick : 29
Timer tick : 30
Timer tick : 31
Timer tick : 32
Timer tick : 33
Timer tick : 34
Timer tick : 35
Timer tick : 36
Timer tick : 37
Entering infinite loop
Timer tick : 38
Timer tick : 39
Timer tick : 40
Timer tick : 41
```

and this goes on forever.... I can't see the GC failing... Please elaborate.

josesimoes, 2018-03-05:
Changing the code to stress the GC with this:

`            strb.AppendLine("TestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTestTest");
`

The GC is called here:

```
Timer tick : 148
Timer tick : 149
GC: 1msec 8424 bytes used, 375804 bytes available
Type 0F (STRING              ):    108 bytes
Type 11 (CLASS               ):    420 bytes
Type 12 (VALUETYPE           ):    144 bytes
Type 13 (SZARRAY             ):    960 bytes
   Type 04 (char             ):    936 bytes
   Type 07 (I4               ):     24 bytes
Type 15 (FREEBLOCK           ): 375804 bytes
Type 17 (ASSEMBLY            ):   4896 bytes
Type 1B (DELEGATE_HEAD       ):     72 bytes
Type 1D (OBJECT_TO_EVENT     ):     24 bytes
Type 1F (THREAD              ):    768 bytes
Type 20 (SUBTHREAD           ):     96 bytes
Type 21 (STACK_FRAME         ):    864 bytes
Type 22 (TIMER_HEAD          ):     72 bytes
Timer tick : 150
Timer tick : 151
Timer tick : 152
Timer tick : 153
```
and again for a number of times:

```
Timer tick : 593
Timer tick : 594
Timer tick : 595
GC: 1msec 8028 bytes used, 376200 bytes available
Type 0F (STRING              ):    120 bytes
Type 11 (CLASS               ):    276 bytes
Type 12 (VALUETYPE           ):    144 bytes
Type 13 (SZARRAY             ):     84 bytes
   Type 04 (char             ):     60 bytes
   Type 07 (I4               ):     24 bytes
Type 15 (FREEBLOCK           ): 376200 bytes
Type 17 (ASSEMBLY            ):   4896 bytes
Type 1B (DELEGATE_HEAD       ):     72 bytes
Type 1D (OBJECT_TO_EVENT     ):     24 bytes
Type 1F (THREAD              ):    768 bytes

Type 20 (SUBTHREAD           ):     96 bytes
Type 21 (STACK_FRAME         ):   1476 bytes
Type 22 (TIMER_HEAD          ):     72 bytes
Timer tick : 596
Timer tick : 597
Timer tick : 598 
```

piwi1263, 2018-03-05:
Try to perform 

 `Console.WriteLine("Available memory : " + Debug.GC(true));` 

Put this line of code at the beginning of main() and it made my f4 discovery pretty undeployable unexplorable

josesimoes, 2018-03-05:
Calling `Debug.GC()` with `true` forces the GC _and_ the heap compactation which is causing issues.

See [here](https://github.com/nanoframework/Samples/commit/36cf6be810ad71ec43fb2e732b620b868566ab39).

This is plaguing nF since the initial import and was never solved...

josesimoes, 2018-03-05:
QUAIL is running GC.Test app (from samples repo) with 0.1.0-preview.555  for 10 minutes now with LED1 blinking.

josesimoes, 2018-04-19:
Blocked by #307.
The heap compaction has to be solved first as it's the most likely cause for this.

josesimoes, 2018-05-18:
@piwi1263 

> `Console.WriteLine("Available memory : " + Debug.GC(true));`

Just found that one can't call `Debug.GC()` with the request to force heap compaction at the very beginning of an app. That causes an immediate hard fault exception. But only with STM32 devices. ESP32 tolerates that very well....

josesimoes, 2018-05-25:
Please provide feedback on this after nanoframework/nf-interpreter#705.

MikroBusNet, 2018-05-25:
```
private static void TestTimer()
        {
            int2.SetDriveMode(GpioPinDriveMode.Output);
            var _timer = new Timer(new TimerCallback(TimerHandler), null, 0, 500);
        }

        private static void TimerHandler(object status)
        {
            var tab = new string[400];
            Hardware.Led1.Toggle();  // Heartbeat
        }
```
GC does seem to kick in as after roughly 5-6 seconds the led is steady. This is not the point here, of course, but it means that the board is "hung" at some point. As expected, would I say.

However, VS still has control and debug session can be stopped and restarted with fresh code.

Same with `Debug.GC(true);` but the board is not hung, here : execution continues after the statement.

MikroBusNet, 2018-05-25:
To be more precise, the board is not hung, in fact.
If I start another thread along the above timer, then only the timer is hung. The other thread continues its execution.

josesimoes, 2018-05-25:
I'm not following you... Could you post the full app so I can run it locally and see whatever is occurring?

MikroBusNet, 2018-05-25:
At some time, the timer will hang. But not the heartbeat. That's why I say that the _board_ is not hung.

```
public static void Main()
{
    try
    {
        WriteLine("Program started");
        WriteLine("Memory available : " + Debug.GC(true));
		
        var _timer = new Timer(new TimerCallback(TimerHandler), null, 0, 500);
        HeartBeat();
    }
    catch (Exception ex) when (Debugger.IsAttached)
    {
        WriteLine("Exception caught : " + ex.Message);
    }
    catch
    {
        while (true)
        {
            Hardware.Led3.Toggle();
            Thread.Sleep(50);
        }
    }
    finally
    {
        WriteLine("Entering infinite loop...");
        while (true)
        {
            Thread.Sleep(100);
        }
    }
}

private static void TimerHandler(object status)
{
    var tab = new string[400];	// To make GC kick in at some time
    Hardware.Led1.Toggle();
}

private static void HeartBeat()
{
    while (true)
    {
        Hardware.Led2.Toggle();
        Thread.Sleep(200);
    }
}
```
What is surprising, however, is that the initial `Debug.GC(true);` statement does not produce a hang although it should force GC to run.
In the timer handler, I presume that some kind of _GC(true)_ is called as well but in this case it stops the execution or goes in an infinite loop, I don't know.

josesimoes, 2018-05-25:
Will take a look at that later.

If you care to take a look I've published in the samples repo my test app to stress the GC.

josesimoes, 2018-05-25:
> What is surprising, however, is that the initial Debug.GC(true); statement does not produce a hang although it should force GC to run.

This is because the original hang was caused by a lack of heap allocation not an issue in the memory compaction code. At least this is the result of my findings. Another explanation for that was the issues with the event scheduler that I've fixed too.

josesimoes, 2018-06-18:
This behaviour doesn't seem to be occurring anymore so closing.

closed 2018-06-18 by josesimoes