nanoframework/Home #291: GPIO Read does return a magical number

piwi1263, 2018-03-03:
## Details about Problem
When doing a GpioPin pin.Read() a magical number is returned instead of a "1" or a "0". This behavior started when flashing a new FW based upon ChibiOS 18.2.x

Using:

nanoFramework extension: 0.3.8.38
GPIO assembly: V1.0.0-preview174
Board: Netduino 3 Wifi

VS version (if appropriate): VS2017 15.5.6
OS version (i.e. win10 v1607 (14393.321)): Win 10 Pro 10.0.16299

Worked before? If so, with which nanoFramework version: Worked before the change to ChibiOS 18.2.x

## Detailed repro steps so we can see the same problem

Project to reproduce:
```
using System;
using System.Threading;
using System.Diagnostics;

using Windows.Devices.Gpio;
using Windows.Devices.I2c;

namespace nd3wifi
{
    public class Program
    {
        #region GPIO
        // GPIO
        private const int GPIO_PIN_PA10 = 0 * 16 + 10;  // Led (Blue) for Netduino3 Wifi
        private static GpioPin _led;
        private static GpioController _gpio;
        private static bool _blinky = false;
        #endregion

        public static void Main()
        {
            try
            {
                // GPIO
                _gpio = GpioController.GetDefault();
                // The led
                _led = _gpio.OpenPin(GPIO_PIN_PA10);
                _led.SetDriveMode(GpioPinDriveMode.Output);
                _led.Write(GpioPinValue.Low);

                // Use a simple system timer to do a blinky
                Timer _seconds = new Timer(Tick, null, 0, 1000);

                // Now what
                for (; ; )
                {
                    Thread.Sleep(50);
                }

            }
            catch (Exception ex)
            {
                // Do whatever please you with the exception caught
            }
        }

        public static void Tick(object info)
        {
            // Assuming the timer event will hit so once in a wile

            GpioPinValue _test = _led.Read();

            GpioPinValue _highValue = GpioPinValue.High;

            GpioPinValue _lowValue = GpioPinValue.Low;

            _led.Write(_led.Read() == GpioPinValue.High ? GpioPinValue.Low : GpioPinValue.High);
        }
    }
}

```
Result:
[image]

Other suggested things
Have a look at the change/commit to the native Gpio code what value is returned. 

Previous code : uint16_t currentValue = palReadPad(port, pad);

Current code: uint16_t currentValue = palReadLine(GetIoLine(pinNumber));

By using a global boolean and coding the old way does blink the LED.

        public static void Tick(object info)
        {
            // Assuming the timer event will hit so once in a wile

            //GpioPinValue test = _led.Read();
            //GpioPinValue _highValue = GpioPinValue.High;
            //GpioPinValue _lowValue = GpioPinValue.Low;

            if (_blinky == true)
            {
                _led.Write(GpioPinValue.Low);
                _blinky = false;
            }
            else
            {
                _led.Write(GpioPinValue.High);
                _blinky = true;
            }

            //_led.Write(_led.Read() == GpioPinValue.High ? GpioPinValue.Low : GpioPinValue.High);
        }

MikroBusNet, 2018-03-03:
Events do not work anymore either. Quite logical, though.

piwi1263, 2018-03-05:
Hi José, this will cover the fix probably. Haven't tested yet. But the next time the GPIO test in the samples repo doesn't cover the GPIO Read functionality.  I'd suggest this line in addition in the for loop with a shorter delay so it can be spotted directly.

// Different timing usage to be able to see the working of the read function on an output port 
_redLED.Write(_redLED.Read() == GpioPinValue.High ? GpioPinValue.Low : GpioPinValue.High);
Thread.Sleep(250);
_redLED.Write(_redLED.Read() == GpioPinValue.High ? GpioPinValue.Low : GpioPinValue.High);
Thread.Sleep(250);

closed 2018-03-05 by piwi1263