nanoframework/Home #302: Missing assemblies deployment

piwi1263, 2018-04-04:
## Details about Problem

When deploying a project that references another project the assemblies, packages of the second (or third or forth) project are not added to the list of asemblies to be deployed.

VS version (if appropriate): Current as of this writing

OS version (i.e. win10 v1607 (14393.321)): N/A

Worked before? If so, with which nanoFrameowrk version: Not detected if ever did work.

## Detailed repro steps so we can see the same problem

1. Create new project and add some package like GPIO or ADC or PWM and some code.

2. Create new project and refer to project in step 1 only. Set project 2 as startup project and start deployment of project created in step 2. The referred assemblies of project one are not in the deployment to the target board.

josesimoes, 2018-04-08:
Seems to be happening

josesimoes, 2018-04-08:
@steffalk is waiting for this too

steffalk, 2018-04-08:
Hello everybody, no urgent problem here. I can wait until this gets fixed and will be happy with it.

josesimoes, 2018-04-09:
It turns out that the deployment provider isn't able to access the resolved assemblies of referenced projects.
This means that the workaround for this is:
- Set the referenced projects to NOT deploy
- Add the referenced assemblies of referenced project in the main app project so they get deployed.

closed 2018-04-09 by josesimoes