nanoframework/Home #304: Program with thight loop disables communication with Netduino 3 board

steffalk, 2018-04-08:
## Details about Problem

Using a nanoframework program with tight loops on my Netduino 3 boards causes the board only be reachable by reflashing the firmware.

## Detailed repro steps so we can see the same problem

1. Example code:

public static class Main
{
public static void Main()
{
while (true) { };
}
}

You can also, for example, read a GPIO port in the loop.

2. Run that program.

3. Try to ping the board or erase the deployment area. The device no longer answers, as it seems because the CPU is so heavily used.

4. Workaround: Insert some Thread.Sleep() in the loop.

josesimoes, 2018-04-09:
This behaviour is by design. As you've figured it out, the CPU is busy running the CLR that it lacks time to properly serve the other RTOS threads (namely the USB driver) which causes the board to become irresponsive.

The "workaround" is actually in the application template provided.
You can take this even further by wrapping the Thread.Sleep() in a compiler define that includes that if you are on debug build and removes it on release build, for example.

steffalk, 2018-04-09:
Thank you for your answer, but there remain questions for my understanding, if you permit:

a) If the OS is a real time OS, and the USB communication runs on a separate OS thread, how can the intended workload "eliminate" the outside communication to the point that the board does not react even to a "ping"?

b) As I can't remember that I have ever seen this behavior running original netmf (4.3), it could improve acceptance for hobbyists like me who make trivial errors (such as a tight loop) and then wonder how they can get the board "useable" again. At least, we should clearly teach them about this behavior, but even better would be if we could improve the board's behavior.

josesimoes, 2018-04-09:
> a) If the OS is a real time OS, and the USB communication runs on a separate OS thread, how can the intended workload "eliminate" the outside communication to the point that the board does not react even to a "ping"?

You are correct about this. The USB driver runs on a RTOS thread that as the same priority as the CLR. This is because the goal is to priviledge the CLR execution over anything else. As the CLR thread eats most of the CPU, timeouts occur in the WireProtocol thread that make the device look irresponsive to VS. Actually it is hapily running the application but it looks like it's broken...

> b) As I can't remember that I have ever seen this behavior running original netmf (4.3), it could improve acceptance for hobbyists like me who make trivial errors (such as a tight loop) and then [...]

I understand what you are saying and that makes sense. It would require some time to experiment the best way to overcome this. I'm reopening the issue and hopefully someone will grab it. 😉

steffalk, 2018-04-26:
Thanks! I think it would be worthy if we always simply can press F5 in VS and the new version of our program simply runs. And if it is found buggy, stop the debugger, change code, F5 again. This is the experience I have from netmf, and users with not so deep knowledge would expect it here also, or even be far more happy.

josesimoes, 2018-05-25:
Increasing the priority of Receiver RTOS thread and increasing it's heap seems to dramatically improve the responsiveness of the target on high CPU apps.

closed 2018-05-25 by josesimoes