nanoframework/Home #315: Native stubs not getting generated for custom nanoFramework libraries

sharmavishnu, 2018-05-01:
Steps to reproduce:
1. Create a new nanoFramework library project
2. Create a class and add a method with [MethodImplAttribute(MethodImplOptions.InternalCall)] attribute
3. Enable "Generate stub files" option and build the project

Result:
No sub files get generated

Observations:
1. If you download and build any of the core libraries from this repository (e.g. nanoFramework.Runtime.Events), the stubs are correctly generated
2. The project file for the core libraries is different from the project file of custom libraries that you create using the VS nanoFramework extension.
3. One needs to select "No (True)" for "Generate stub files" for enabling stub file generation. This is confusing. Needs to be corrected.

Environment:
1. Windows 10 PC with Visual Studio 2017 and nanoFramework extension 0.3.8.51

sharmavishnu, 2018-05-04:
Checked the build log and I see this:
1>  Done executing task "MetaDataProcessorTask".
1>  Task "CallTarget" skipped, due to false condition; ('$(NF_GenerateStubs)'=='true') was evaluated as (''=='true').

The $(NF_GenerateStubs) is not getting set for some reason.

josesimoes, 2018-05-04:
That's exactly what I saw when confirming the bug. You're on it and working on a fix? 😉

sharmavishnu, 2018-05-04:
I have been struggling since morning to get my build environment right to look at this bug, but no success as yet. Keep getting one build issue after another. 

Looking at the source code so far, here is my understanding:
1. It seems, in the file NFProjectSystem.MDP.targets, line 192 with the statement 
<CallTarget Targets="GenerateStubs" Condition="'$(NF_GenerateStubs)'=='true'" /> fails, because $(NF_GenerateStubs) is never getting set. 
2. It does not seem like that $(NF_GenerateStubs) is getting set anywhere
3. If we look at the UI (xaml), the property that gets set is NFMDP_STUB_SKIP

So just asking, shouldn't the line 192 in file NFProjectSystem.MDP.targets be

<CallTarget Targets="GenerateStubs" Condition="'$(NFMDP_STUB_SKIP)'=='true'" />

MikroBusNet, 2018-05-05:
If you did not already, you should have a look a class libraries nfproj files as references. You will see something like the following (exceprt from i2c lib) :
```
<PropertyGroup>
    <!-- override default options for MetaDataProcessor -->
    <!-- generate PE: TRUE -->
    <NFMDP_PE_SKIP>false</NFMDP_PE_SKIP>
    <NFMDP_PE_Verbose>true</NFMDP_PE_Verbose>
    <NFMDP_PE_VerboseMinimize>true</NFMDP_PE_VerboseMinimize>
    <!-- generate STUBS: TRUE -->
    <NFMDP_STUB_SKIP>false</NFMDP_STUB_SKIP>
    <NFMDP_STUB_Verbose>true</NFMDP_STUB_Verbose>
    <!-- this is one is absolutely mandatory for base class libraries -->
    <NFMDP_STUB_SkeletonWithoutInterop>true</NFMDP_STUB_SkeletonWithoutInterop>
    <NFMDP_STUB_VerboseMinimize>true</NFMDP_STUB_VerboseMinimize>
    <NFMDP_STUB_GenerateSkeletonFile>Stubs\win_dev_i2c_native</NFMDP_STUB_GenerateSkeletonFile>
    <NFMDP_STUB_GenerateSkeletonProject>win_dev_i2c</NFMDP_STUB_GenerateSkeletonProject>
    <NFMDP_CMD_LINE_OUTPUT>false</NFMDP_CMD_LINE_OUTPUT>
  </PropertyGroup>
```
You will have to manually add those lines to your own nfproj file and of course edit some parts to reflect your project name.
Not sure if it will be enough but it should be a good start anyway.

sharmavishnu, 2018-05-06:
My earlier assessment about `<CallTarget Targets="GenerateStubs" Condition="'$(NF_GenerateStubs)'=='true'" />` line with NF_GenerateStubs not getting set was not correct. Even for the core libraries, where the stubs are getting created, the condition is false because even in that case, the NF_GenerateStubs is empty. So it seems, that line does nothing.

Based on feedback from @MikroBusNet , it seems, the only workaround right now, is to manually update the project files (or, create a copy of a library file from the repository and simply rename and re-configure). This workaround creates the needed stub files.

So, it seems the issue is with the template.

closed 2018-06-14 by josesimoes