nanoframework/Home #316: Console.WriteLine() does not write text in the debugger's output window

MikroBusNet, 2018-05-04:
## Details about Problem
With latest firmware for both Quail and F429iDiscovery, _Console.WriteLine()_ does not output anything in the output windows of VS debugger.

VS version : 15.6.6
nf extension : 0.3.8.57
Quail firmware : preview614
F429iDiscovery firmware : preview615

## Detailed repro steps so we can see the same problem
```
public static void Main()
        {
            try
            {
                while (!Debugger.IsAttached) { Thread.Sleep(100); }
                Console.WriteLine("Program started");

                Thread.Sleep(1000);
            }
            catch (Exception ex)
            {
                Console.WriteLine("Exception : " + ex.Message);
            }
            finally
            {
                Console.WriteLine("Entering infinite loop");
                while (true)
                {
                    Thread.Sleep(100);
                }
            }
        }
```
Both Quail and F429 won't show anything in the output window. However, F429 is showing the texts in the SWO output.

[image]

ishvedov, 2018-05-06:
+ESP32 starting from 600+ firmware.

josesimoes, 2018-05-06:
@ishvedov interesting that. To clarify: Using the same VS extension  and referring the image to a version earlier than 600, console output works? Is this correct?

Btw, which VS extension version are you in?

piwi1263, 2018-05-06:
At the moment this is for various community boards sofar, if not all.

Using VS version 15.6.7 and extension 0.3.8.57.

ishvedov, 2018-05-08:
Using VS 15.6.7 and 15.7, extension 0.3.8.57.

ESP32:
fw 592 working
fw 602+ not working

ishvedov, 2018-05-09:
VS 15.7.1, fw 592 print only NFPKTV1 at the beginning.

josesimoes, 2018-05-09:
That's the packet header. It shouldn't be printed... 😯 
Anyone working with a STM32 target has noticed this?

ishvedov, 2018-05-10:
VS 15.7.1, Extension 0.3.8.58, fw 617 - not working.

josesimoes, 2018-05-10:
@ishvedov if you can try loading the fw of nanoframework/nf-interpreter#681 and give it a try.
VS extension v0.3.8.58 is mandatory.

ishvedov, 2018-05-11:
@josesimoes I try this...
`System Information
HAL build info: 0.1.0.681, nanoFramework running ESP32 built with IDF v3.0-rc1 
Image build @ May 10 2018 GNU ARM GCC v5.2.0
OEM Product codes (vendor, model, SKU): 0, 0, 0
Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000
Solution Build Info: 0.1.0.681, nanoFramework running ESP32 built with IDF v3.0-rc1 
AppDomains:
Assemblies:`
Time to time I have different errors:
- >  The nanoDevice target and the debugger engine failed to initialize because of unspecified device errors. The debugger engine thread has terminated unexpectedly with error 'Device is reporting no assemblies deployed. Can't debug!'.
- MsgBox with error 0x80004005
- > Error: a2000000 Waiting for debug commands... The nanoDevice runtime is loading the application assemblies and starting execution.
Also can't deploy at most times. I can run only one empty project with Console.WriteLine via F10, with F5 I can't do it. And so on. Like early 500 builds. With simple project only one run, I saw output from Console.WriteLine.

MikroBusNet, 2018-05-11:
Latest alpha-681 does write text in the VS output window. At least with Quail for the moment.

josesimoes, 2018-05-11:
Please give it a try with VS extension v0.3.8.61 **BUT USING THE REGULAR FW IMAGE** (forget the nanoframework/nf-interpreter#681 for the moment).

MikroBusNet, 2018-05-11:
This one is working as expected, with fw 617 on Quail.
It even looks like it is "faster" to start debug session...

MikroBusNet, 2018-05-11:
It seems that it also resolves #317

josesimoes, 2018-05-11:
Confirmed by several folks for both STM32 and ESP32 targets.

Fixed with nanoframework/nf-Visual-Studio-extension#231. (VS extension v0.3.8.61)

closed 2018-05-11 by josesimoes

mesut25, 2022-10-07:
nanoff --update --target ESP32_WROOM_32 --serialport COM6

no support for booting. I downloaded from here, but I am getting an error. Device appears, but there is no flashing error, no output.

mesut25, 2022-10-07:
https://docs.nanoframework.net/content/getting-started-guides/getting-started-managed.html

I SUCCESS. from here  
nanoff --platform esp32 --serialport COM6 --update