nanoframework/Home #319: I2C does not send data

MikroBusNet, 2018-05-07:
## Details about Problem

When writing to an eeprom (I2C device), no data is written (resulting in no data read in this case).
The SDA line does not show any activity.

Quail firmware : preview617
Worked before? Yes, but can't remember which firmware version was ok.

## Detailed repro steps so we can see the same problem
The following code was working "before" and is currently working _as is_ on the same board with TinyCLR.

```
public override void ReadData(Int32 address, Byte[] data, Int32 index, Int32 count)
{
  _eeprom.Write(new[] { (Byte)(address >> 8), (Byte)(address & 0xFF) });
  Thread.Sleep(5);
  var buffer = new Byte[count];
  _eeprom.Read(buffer);
  Array.Copy(buffer, 0, data, index, count);
}
```
### Screenshot (if applicable/relevant)
Screenshot using nanoFramework :
[image]

Screenshot using TinyCLR :
[image]

MikroBusNet, 2018-05-11:
On the F429 with fw617, a _Read_ command is issued after a _Write_, which is not expected :

[image]

The code is this one :
```
public override void WriteData(Int32 address, Byte[] data, Int32 index, Int32 count)
        {
            var buffer = new Byte[data.Length + 2];
            buffer[0] = (Byte)(address >> 8);
            buffer[1] = (Byte)(address & 0xFF);

            Array.Copy(data, index, buffer, 2, count);
            _eeprom.Write(buffer);
            Thread.Sleep(5);
        }
```
The call is done this way : `WriteData(0, new[] { 200 }, 0, 1);`

Same for readings, there seem to be "duplicated" buffers or something like that. Here is the result for a single _Read_ command :
[image]
The normal sequence should have been the first "Setup Write" and "Setup Read". The other one has not been sent.

MikroBusNet, 2018-05-11:
Same behaviour on Quail, F429iDiscovery and Nucleo F746ZG.

MikroBusNet, 2018-05-13:
I would now add another issue raised by Stefan Falk on Slack : a simple _Write()_ operation is taking between 8 and 11 ms to achieve !
More precisely : it takes 10 ms for the managed app to get control back. The timings on I²C line is correct, given the frequency used. So this is not the transfer itself that takes time but rather code around it.
SPI does not show this behaviour at all. Granted, SPI is faster than I²C but this does not explain this huge difference, to me. It takes much less than 1ms for SPI with the same data size sent.

josesimoes, 2018-05-13:
That's explained by the new workflow with CLR events.
The SPI implementation has an "improved" algorithm that figures out if the transaction _fits_ in thread execution slot. If it does it will skip setting up the continuation and event. 
This same algorithm is still to be ported to the I2C code.

MikroBusNet, 2018-05-13:
I don't think that this applies, here. The 10ms timing is for 2 bytes sent... If 2 bytes don't fit in an execution slot, then we have another problem.

josesimoes, 2018-05-13:
Please read again the above.

The smart algorithm is **NOT** implemented in the I2C code. Yet.
That means that it doesn't matter if the buffer is 2 byte or 2000 bytes: it will take the same time to start the transaction.
Plus ChibiOS I2C async API works differently of the other drivers there: it requires a thread on its own. Setting up that thread and starting it requires relinquish control from the CLR ChibiOS thread and waiting for that I2C to get it's execution time.
All this accounts for what you guys are seeing here.

If you have a different understanding please point the code line (or code block) that you see that is causing this behaviour.

MikroBusNet, 2018-05-13:
What I understand here is that when the "smart algorithm" will be coded, we will have such a 10ms penalty if the code decides that the data to be transferred won't fit in an execution slot.
Am I right ?

josesimoes, 2018-05-13:
The penalty will depend on cpu clock and execution time, but yes, it will exist.

josesimoes, 2018-05-16:
Just found root cause for those 10ms between thread re-scheduling that weren't there when you've tested the previous implementation:

1) Fix to events [here](https://github.com/nanoframework/nf-interpreter/blame/6a37b658060fd8dee49bc703865471ff06e92f9e/targets/CMSIS-OS/ChibiOS/nanoCLR/targetPAL_Events.cpp#L127)

2) And later [here](https://github.com/nanoframework/nf-interpreter/blame/4bab57535428a00a3acbd380eaec0cf2831eccdc/targets/CMSIS-OS/ChibiOS/nanoCLR/targetPAL_Events.cpp#L150)

As you can see it's not caused by the recent changes in the code.

and

closed 2018-05-24 by josesimoes