nanoframework/Home #320: Program hangs when using non-static arrays

MikroBusNet, 2018-05-10:
## Details about Problem
A very simple program using arrays inside methods is hanging after short time.
It looks like it is related to GC.

Quail firmware : preview617

### Sample Project (if applicable)
```
using System;
using System.Threading;
using System.Diagnostics;
using Windows.Devices.Gpio;

namespace TestGC
{
    public class Program
    {
        private static GpioPin Led1, Led3;
        //private static int[] tab = new int[400];

        public static void Main()
        {
            try
            {
                Init();

                new Thread(Thread1).Start();
            }
            catch (Exception ex) when (Debugger.IsAttached)
            {
                Console.WriteLine("Exception caught : " + ex.Message);
            }
            catch
            {
                while (true)
                {
                    Led3.Write(GpioPinValue.High ^ Led3.Read());
                    Thread.Sleep(50);
                }
            }
            finally
            {
                while (true)
                {
                    Thread.Sleep(100);
                }
            }
        }

        private static void Thread1()
        {
            while (true)
            {
                Led1.Write(GpioPinValue.High ^ Led1.Read());
                int[] tab = new int[400];
                for (var i=0; i<400; i++)
                {
                    tab[i] = i;
                }
                Thread.Sleep(50);
            }
        }

        private static void Init()
        {
            var gpioController = GpioController.GetDefault();
            Led1 = gpioController.OpenPin(4 * 16 + 15);
            Led1.SetDriveMode(GpioPinDriveMode.Output);
            Led1.Write(GpioPinValue.Low);
            Led3 = gpioController.OpenPin(2 * 16 + 3);
            Led3.SetDriveMode(GpioPinDriveMode.Output);
            Led3.Write(GpioPinValue.Low);
        }
    }
}
```
The program above runs about 1 min before it hangs. If you uncomment the static array declaration (and comment the one in the thread method) then it is running fine. At least for 40 minutes as of writing.

josesimoes, 2018-05-10:
This is a good sample project to reproduce the GC issue. Thanks for providing it.

Closing as this is a duplicate of #290.

closed 2018-05-10 by josesimoes

MikroBusNet, 2018-05-10:
The above statement (mine) can be misleading : there is not absolute need to have a static variable to avoid the hang.
It is more the constant creation of the array, in this sample, that is the cause of the GC kicking in. If you move the array declaration that is inside the thread method just before the _while-loop_, then there is no problem either, the board is running for hours.

```
private static void Thread1()
        {
            int[] tab = new int[400];
            while (true)
            {
                Led1.Write(GpioPinValue.High ^ Led1.Read());
                //int[] tab = new int[400];
                for (var i=0; i<400; i++)
                {
                    tab[i] = i;
                }
                Thread.Sleep(50);
            }
        }
```