nanoframework/Home #325: Thread.Sleep() is not accurate

MikroBusNet, 2018-05-17:
## Details about Problem

I don't expect sub-millisecond accuracy but there is something wrong, here.

With the code below, I would expect a 90ms period with a pulse width of 45ms.

Tested on Quail with firmware 625.
```
private static void TestSleep()
        {
            var int2 = Hardware.gpioController.OpenPin(Hardware.SocketTwo.Int);
            int2.SetDriveMode(GpioPinDriveMode.Output);
            while (true)
            {
                int2.Write(GpioPinValue.High);
                Thread.Sleep(45);
                int2.Write(GpioPinValue.Low);
                Thread.Sleep(45);
            }
        }
```

### Screenshot

Very helpful if you send along a few screenshots to help visualize the issue!
[image]

josesimoes, 2018-05-17:
This is the best it can do:

[image]

josesimoes, 2018-05-17:
Fixed with nanoframework/nf-interpreter#690.

MikroBusNet, 2018-05-17:
I am sorry to say that it is not fixed with fw 628 :(

There is something else because if you take multiples of 10 for the sleep value, then you have accurate timings. But as soon as you chose something else, like 12, 23, 37, 121 then you always get the value rounded up to the next multiple of 10.
With the value given just above, you would get 20, 30, 40 and 120 ms real sleep time. And as accurate as it can be for those values, which is something around 0.1 to 0.2 ms. This is not bad and is perfectly acceptable.

If you choose multiples of 10, then you get the correct and accurate timings. With the same accuracy of 0.1 to 0.2 ms

So I reopen this issue.

steffalk, 2018-05-19:
The described behavior does not sound unexpected to me: Sleep() sleeps at least the given amount of time, and the thread then will continue at the next regular context switch, which occurs each 10 ms. Eventually, that could be improved (for cases where there is only one thread), but I would not go so far and call this a bug. Just my 2 Cent.

MikroBusNet, 2018-05-19:
Here is a summary of my last measurements on this. The test has consisted in measuring the pulse width on the oscilloscope, for sleeps from 1ms to 10ms, and then for 20ms and 30ms.
I have tested with nanoFramework, TinyCLR and NETMF 4.4 (MBN port) :
 - NETMF : every sleep time was accurate between 0.08ms and 0.1ms drifts (1.08 ms measured for 1 ms requested, 10.1ms measured for 10ms requested)
 - TinyCLR : every sleep time was accurate between 0.188ms and 0.196ms drifts (1.188 ms measured for 1 ms requested, 10.196ms measured for 10ms requested)
 - nanoFramework : only multiples of 10ms were accurate between 0.08ms and 0.20ms drifts (10.08 ms measured for 1 ms requested, 10.20ms measured for 10ms requested). For all non-multiples of 10,the result was rounded up to the next multiple of 10, the accuracy being the same.

Once the test has been launched, I did let it run about 30 seconds for the same frequency and looked at the variation of the pulse width.
- NETMF was very stable with an almost constant 0.1 ms difference between requested and measured pulse width.
- nanoFramework and TinyCLR were less stable regarding the drift but the change was not present that often. But still clearly noticeable compared to NETMF.

closed 2018-05-24 by josesimoes

MikroBusNet, 2018-05-25:
Side notes on this one since the PR :
 - pulse width is now correct for every requested timing
 - accuracy is 0.1 ms almost constant with a loop writing twice to gpio
 - accuracy is +/- 0.004 ms when using a _toggle()_
 - max drift is now +/- 0.004 ms in all cases

MikroBusNet, 2018-05-27:
On Nucleo-F746ZG, timings are far less accurate :(
I have always an almost 0,14ms lower pulse width than the one requested : 6,86ms for 7ms requested, for example.
Both _Thread.Sleep()_ and _Timer()_ are impacted the same way. This is using pin _Toggle()_ or classic high/low loop.

I remember having those unaccurate timings a long time ago on F7 boards and can't remember if they were solved.
This does not question what has been done here. It's only that there is perhaps something else to figure out with F7 boards about timings.

I don't re-open this issue for this, however. If needed, a dedicated one will be created.