nanoframework/Home #327: Getting 'System.ArgumentException' Error while calling  'SpiDevice.FromId'

thambirm, 2018-05-23:
## Details about Problem

nanoFramework area : C# code | Hardware/target board 

VS version (if appropriate): VS 2017

OS version : Windows 10 Pro (1709)

Target Board : ESP32 DevCKit

Worked before? If so, with which nanoFrameowrk version: No

## Detailed repro steps so we can see the same problem

1. Initializing SPI device failing 
**SpiDevice.FromId("SPI1", new SpiConnectionSettings(22));**

2.Getting 'System.ArgumentException' type Error Message 

3. Tried changing the Bug Number and CS PIN number , but results are same .
...

## Other suggested things

### Sample Project (if applicable)

Very helpful if you can zip a project and paste into this issue!

### Screenshot (if applicable/relevant)

Very helpful if you send along a few screenshots to help visualize the issue!
[image]

josesimoes, 2018-05-23:
Can this be related with #323? 
Does this occur when you are instantiating a second SPIDevice?

MatthiasJentsch, 2018-05-27:
I can reproduce the issue. It occurs when initializing the first device. Internally we call the ESP-IDF functions spi_bus_initialize and spi_bus_add_device. With "SPI1" you configure the HSPI bus. We use the pins GPIO_NUM_23 for MOSI, GPIO_NUM_25 for MISO and GPIO_NUM_19 for CLOCK. The CS pin number is defined via SpiConnectionSettings. I the example it is pin 22.
The reason why this results in an ArgumentException is most likely caused by returning not ESP_OK from the spi_bus_add_device function. I'll do some more investigations why this is failing.

MatthiasJentsch, 2018-05-28:
The spi_bus_add_device function fails because the clock frequency is unset. You should set the clock frequency. Example for 1MHz clock frequency:
```
SpiConnectionSettings settings = new SpiConnectionSettings(22);
settings.ClockFrequency = 1000000;
SpiDevice.FromId("SPI1", settings);
```
This code runs without an exception.
With the example above you initialize a SPI device which are using the HSPI bus and the following pins:
* GPIO_NUM_23 for MOSI
* GPIO_NUM_25 for MISO
* GPIO_NUM_19 for CLK
* GPIO_NUM_22 for CS

MatthiasJentsch, 2018-05-29:
If the developer doesn't set a frequency we should assume that the max frequency will be used.
I'd like to correct this behavior and would set it to max frequency (26 MHz) if the developer doesn't set a frequency.

josesimoes, 2018-05-29:
Fixed with nanoframework/nf-interpreter#711.

closed 2018-05-29 by josesimoes