nanoframework/Home #332: Visual Studio freezes and crash on start-up after installing nanoFramework extension

trabuzin, 2018-05-26:
## Details about Problem

nanoFramework area Visual Studio extension

VS version (if appropriate): irrelevant, same problem was manifesting with other versions too

OS version (i.e. win10 v1607 (14393.321)): irrelevant

## Detailed repro steps so we can see the same problem

After installing VS extension, VS would freeze and crash after some time on start-up. With help of people from support-user-question Slack group, we discovered that extension would enumerate ports on start-up as can be seen in: 
[nanoFramework_VS_crash.txt](https://github.com/nanoframework/Home/files/2041542/nanoFramework_VS_crash.txt)

Further investigation showed that I had two Standard Serial over Bluetooth link ports (COM4 and 5) in Ports (COM&LPT) section of device manager
[image]

Disabling built-in Bluetooth device through BIOS removed these two ports and resolved problem with freezing and crashing of VS. Enabling it again would, of course, make VS freeze and crash, so these ports are the most probable reason.

The device in question
[image]
Hardware ids of Bluetooth device:
USB\VID_0A5C&PID_21E1&REV_0112
USB\VID_0A5C&PID_21E1

Hardvare ids of Standard Serial over Bluetooth links:
BTHENUM\{00001101-0000-1000-8000-00805f9b34fb}_LOCALMFG&0000 (COM4)
BTHENUM\{00001101-0000-1000-8000-00805f9b34fb}_LOCALMFG&000f (COM5)

Best regards,
Thomir Rabuzin

josesimoes, 2018-05-27:
@trabuzin please try to update your VS extension to v0.3.8.64 from MyGet. After doing that, enable the Bluetooth devices and check that VS can start as expected.

Waiting for your feedback!

trabuzin, 2018-05-27:
@josesimoes VS still freezes after updating extension to v0.3.8.64 from MyGet. This time I dug a little bit deeper by disabling not whole Bluetooth device through BIOS, but by disabling COM4 and 5 one by one through device manager. VS is not freezing when I disable COM5 leaving only COM4 active. If I disable COM4, leaving COM5 active, then VS would freeze.
Errors in event viewer are the same as in original report.

```
Application: devenv.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.Exception
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at nanoFramework.Tools.Debugger.Serial.EventHandlerForSerialDevice+<OpenDeviceAsync>d__6.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at nanoFramework.Tools.Debugger.PortSerial.SerialPort+<ConnectSerialDeviceAsync>d__29.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at nanoFramework.Tools.Debugger.PortSerial.SerialPort+<ConnectDeviceAsync>d__28.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at nanoFramework.Tools.Debugger.PortSerial.SerialPort+<OnDeviceEnumerationCompleteAsync>d__21.MoveNext()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_1(System.Object)
   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()
```

Seems like this specific one is the culprit:
Standard Serial over Bluetooth link (COM5)
Hardware Ids:
BTHENUM\{00001101-0000-1000-8000-00805f9b34fb}_LOCALMFG&000f

Best regards,
Tihomir Rabuzin

josesimoes, 2018-05-27:
Understood. 
Please provide the full hardware IDs of those two COM ports.
There must be some diference among them.

trabuzin, 2018-05-27:
Yes there is, the last character is different

COM4
BTHENUM\{00001101-0000-1000-8000-00805f9b34fb}_LOCALMFG&0000

COM5
BTHENUM\{00001101-0000-1000-8000-00805f9b34fb}_LOCALMFG&000f

josesimoes, 2018-06-05:
Moving this to backlog as it's affecting only one reported user. Plus we are waiting for feedback for sometime now.

josesimoes, 2018-06-13:
Closing this as there was no feedback from @trabuzin for a long time.

josesimoes, 2018-07-16:
Another developer has reported that it's Bluetooth virtual COM ports are also blocking the extension.
PID & VID are BTHENUM\{00001101-0000-1000-8000-00805f9b34fb}_VID&00010057_PID&0023

closed 2018-07-20 by josesimoes