nanoframework/Home #334: Possible buffer corruption and mishandling with large TX bufers on Serial

josesimoes, 2018-06-02:
As described [here](https://github.com/nanoframework/lib-Windows.Devices.SerialCommunication/pull/19#issuecomment-394081397)

Reported by @MikroBusNet . Thanks for finding and reporting it. 😃  👍

josesimoes, 2018-06-04:
Can you post the code that's being called on the various `display.` calls?
Would like to understand if this is a result of the sequence on the calls that write to the WriteBuffer and/or Store().

MikroBusNet, 2018-06-08:
```
private static SmartGpuReponse SendCommand(byte[] command, bool waitForReply = false)
        {
            _outputDataWriter.WriteBytes(command);
            _outputDataWriter.Store();
            if (!waitForReply) return SmartGpuReponse.Ok;
            while (_inputDataReader.Load(1) < 1) Thread.Sleep(1);

            return (SmartGpuReponse) _inputDataReader.ReadByte();
        }

        private static Byte[] SendCommand(Byte[] command, UInt32 expectedLenght)
	    {
	    _outputDataWriter.WriteBytes(command);
            _outputDataWriter.Store();

            Byte[] responseBuffer = new Byte[expectedLenght];
            
           while (_inputDataReader.Load(expectedLenght) < expectedLenght) Thread.Sleep(1);
           _inputDataReader.ReadBytes(responseBuffer);
		return responseBuffer;
	}
```

josesimoes, 2018-10-13:
Caused by read timeout not being set. Fixed with nanoframework/nf-interpreter#906

closed 2018-10-13 by josesimoes

MikroBusNet, 2018-10-13:
Both read and write timeouts were set before use.

They were set because at the time I wrote this driver, there was an issue with (at least) the read timeout.

Here is the constructor for the display driver :

```
public SmartGpu2(Hardware.Socket socket, DisplayType displayType, Boolean showSplashScreen = false)
        {
            _resetPin = GpioController.GetDefault().OpenPin(socket.Rst);
            _resetPin.SetDriveMode(GpioPinDriveMode.Output);
            _resetPin.Write(GpioPinValue.Low);

            if (showSplashScreen) Thread.Sleep(10000);

            _resetPin.Write(GpioPinValue.Low);
            Thread.Sleep(100);
            _resetPin.Write(GpioPinValue.High);
            Thread.Sleep(300);

            _serial = SerialDevice.FromId(socket.ComPort);

            // set parameters
            _serial.BaudRate = 9600;
            _serial.Parity = SerialParity.None;
            _serial.StopBits = SerialStopBitCount.One;
            _serial.Handshake = SerialHandshake.None;
            _serial.DataBits = 8;

            // Because we are reading from the UART it's recommended to set a read timeout,
            // otherwise, the reading operation doesn't return until the requested number of bytes have been read
            _serial.ReadTimeout = new TimeSpan(0, 0, 0, 0, 150);
            _serial.WriteTimeout = new TimeSpan(0, 0, 0, 0, 150);

            // Setup data writer for Serial Device output stream
            _outputDataWriter = new DataWriter(_serial.OutputStream);
            // Setup data read for Serial Device input stream
            _inputDataReader = new DataReader(_serial.InputStream);
```

josesimoes, 2018-10-13:
understood. Nevertheless please give it a try when you have the opportunity.