nanoframework/Home #336: Add equivalent of NETMF Spot.Hardware.Utility assembly

MikroBusNet, 2018-06-02:
## Details about Problem

The Microsoft.SPOT.Hardware.Utility contains very useful methods to work with arrays and the _GetMachineTime()_ method that is the equivalent of a UpTime() method.

Those are missing in nf at the moment.

For example, _ExtractRangeFromArray()_ and _CombineArrays()_ allow getting an array without having to create it first. 
The standard _Array.Copy()_ method that could be used instead needs the array to be created first in the managed code, which means that you have to know its length, which is not always possible, for example when you receive data in chunks from serial stream or from network.

josesimoes, 2018-06-04:
Yes, this assembly was never added because most of their API didn't make sense in nanoFramework.

Yes, `GetMachineTime()` can be useful. All the other _time_ methods don't' make sense anymore.

Regarding the receiving of data chunks: this is now handled in a different way with Streams and Readers which take care of the heavy lifting in the background. So you shouldn't be needing those for that.

As for the Array handling methods... I can see the usefulness on some of them... but for very, very specific situations... and all of them can be carried out with what is already there... so we would be committing more flash and ram space to provide these... 

Let's hear from other folks about this.

MikroBusNet, 2018-06-07:
No feedback = no usefulness. Closing.

closed 2018-06-07 by MikroBusNet