nanoframework/Home #342: MBN_QUAIL reports no native assemblies deployed after debug session start-stop

josesimoes, 2018-06-11:
## Details about Problem

nanoFramework area:  Hardware/target board 
This seems to be affecting only this particular target.

## Detailed repro steps so we can see the same problem

1. Start debug session 

2. Stop debug session

3. Start debug session again

Fails here reporting that there are no native assemblies deployed.

Requesting device info also shows no native assemblies present.

josesimoes, 2018-06-11:
This is happening too with other STM32 targets. 
On the seconds debug start the heap fails to allocate memory for the reply with the native assemblies list, thus failing the deployment checks which eventually cause the deployment to fail.

It seems that this is caused by a short CRT heap on some devices along with memory leaks that are showing after stopping a debug session.

MikroBusNet, 2018-06-11:
Could it be because the reply size has grown over time ?

closed 2018-06-12 by josesimoes