nanoframework/Home #343: Broken String.this[]

ishvedov, 2018-06-13:
## Details about Problem

nanoFramework area: interpreter

Worked before? Yes, old firmware's. At least ESP32 668.

## Detailed repro steps so we can see the same problem
```csharp
Console.WriteLine("a\nb"[1].ToString());
```
return 'b', must '\\'.

MikroBusNet, 2018-06-13:
Can't check myself but waht about adding a '@' in front of the string : `Console.WriteLine(@"a\nb"[1].ToString());`
This would not solve the issue but could be a temporary workaround ?

Also, how does this behave ? `Console.WriteLine("a\\nb"[1].ToString());`
(again, I can't test myslef, I am away from home this week)

ishvedov, 2018-06-13:
@MikroBusNet no workaround possible, because on this functionality based PR for CoreLib.