nanoframework/Home #357: Change AppVeyor yaml to separate build of booter and CLR

josesimoes, 2018-07-16:
No need to keep building and releasing booter versions when nothing has changed. Currently nanoBooter is always build along with the nanoCLR which is a waste of time and computing resource. Not to mention that is misleading because you actually don't know if there were any changes or not.

networkfusion, 2019-01-02:
We no longer use appveyor. Is this still applicable?

Ellerbach, 2023-04-18:
@josesimoes is this still needed? I feel the build system has drastically evolved since 2019 :-)

josesimoes, 2023-04-19:
Yes. Despite all that evolution having separate build for the booter still makes sense.

josesimoes, 2026-10-01:
This is solved with distinct versions for CLR and booter. 
A new version is published only if there are actual changes, which pretty much is equivalent to this.

closed 2026-10-01 by josesimoes