nanoframework/Home #359: Improve referenced projects processing to have a correct assemblies list when deploying

josesimoes, 2018-07-16:
There are currently issues when there is a solution with multiple projects, ones referencing the others.

- [ ] the deployment occurs multiple times (need to investigate if it's possible to control the deploy option for each individual project.
- [ ] the references aren't computed correctly (need to improve this, possibly using the ResolveRuntimeDependenciesTask

josesimoes, 2018-07-17:
The current implementation seems to be the best possible with what VS extension offers today.
When working with multiple projects in a solution, even if there are transversed references, all required references have to be present in the executable project.

Also it's worth noting that on a Solution with multiple projects, the developer has to go to configuration manager and uncheck the deploy option on all project but the executable one. The deploy provider takes care of deploying all the required assemblies (providing that they are referenced in the executable project, of course).

steffalk, 2018-07-17:
So, that I understand correctly: If Main references A, and A reference B, we continue to have Main directly reference B also? Still just for my understanding: This is the best that a VS extension can do? Aren't desktop C# and VB nothing else than an extension, and can resolve indirect references fine?

josesimoes, 2018-07-17:
> This is the best that a VS extension can do? Aren't desktop C# and VB nothing else than an extension, and can resolve indirect references fine?

The Msbuild task responsible for this is ResolveAssemblyReference (which is working perfectly fine, btw).
The thing here is that we don't have direct access to it's outputs when needed (that is in the deployment provider).
Right now the code is doing pretty much the same task but just doing the transverse for one level. The Msbuild task is coded to work to any number of indirect references. 
We can change our code to do that too, but it doesn't make much sense to expect that a nanoFramework project has more than one level of reference...

@steffalk here's a challenge: take a look at this algorithm and improve it to add, say, a couple more levels of transverse referencing. It's not complicated and you'll understand it very easily!

The interesting part starts about [here](https://github.com/nanoframework/nf-Visual-Studio-extension/blob/b6b77c7217c13e4eacb4c9ecaa215dc999959436/source/VisualStudio.Extension/DeployProvider/DeployProvider.cs#L160) and follows. 

Feel free to ask any questions and let me know if I should reopen the issue and assigning it you you...  😉

closed 2018-07-24 by josesimoes