nanoframework/Home #360: Update booter code to remove separate blinker thread

josesimoes, 2018-07-16:
Currently nanoBooter is launching a separate thread to blink an LED. There is no need to do that as the blink loop can be in main. This saves some flash and RAM which is good considering that the booter should be kept at the minimum possible size.

MatthiasJentsch, 2018-07-16:
Should we launch the blink loop in main if there is no executable found? Or should we remove that at all?

josesimoes, 2018-07-16:
We should keep the blink... blinking... when there is no valid CLR.
The idea is to remove the thread not the functionality.

I have a working code [here](https://github.com/Eclo/nf-interpreter/blob/add-support-STM32L0-series/targets/CMSIS-OS/ChibiOS/ST_B_L072Z_LRWAN1/nanoBooter/main.c) for a new target I've been working on. Basically what's needed is to copy what is in the thread loop to the main loop. 😉

closed 2018-07-17 by josesimoes