nanoframework/Home #364: i2c Read not working with EEprom

Dweaver309, 2018-07-17:
Please read the following information before posting the issue.

## Before posting the issue...

If you need clarification on something, need help on a particular situation or want to start a discussion, instead of opening an issue it is best to start a conversation on one of our Slack channels. 
Go there and select the channel that's most appropriate to the context so subject matter experts are most likely to answer promptly.
You can join our lively Slack community by filling in this invite form [here](https://nanoframework.wordpress.com/slack-invite-form)

Remove the content above here and fill out details below.

## Details about Problem

nanoFramework area (C# code | Visual Studio extension | Hardware/target board | Nuget packages | Community targets )

VS version (if appropriate):

OS version (i.e. win10 v1607 (14393.321)):

Worked before? If so, with which nanoFrameowrk version:

## Detailed repro steps so we can see the same problem

1.

2.

...

## Other suggested things

### Sample Project (if applicable)

Very helpful if you can zip a project and paste into this issue!

### Screenshot (if applicable/relevant)

Very helpful if you send along a few screenshots to help visualize the issue!

Dweaver309, 2018-07-17:
[EEprom.driver.zip](https://github.com/nanoframework/Home/files/2202508/EEprom.driver.zip)

More information:
https://nanoframework.slack.com/messages/C4MGGBH1P/convo/C4MGGBH1P-1531755978.000562/

Dweaver309, 2018-07-17:
This code in .netmf is working:
https://github.com/Dweaver309/i2cBus/blob/master/i2cBusSourceCode/CSharp/Program.cs

AdrianSoundy, 2018-07-17:
Had a look at the datasheet and the provided code.

The writebyte() function looks ok but the readbyte() is wrong.
The Read random has to be done as 2 operations:-
1) Set the Address
2) Read next byte(s)
Seen in this from datasheet as it has to start bytes.

[image]

So the code for ReadByte needs to be:-

EEprom.Write(new[] { (Byte)(Address >> 8), (Byte)(Address & 0xFF) });
EEprom.Read(Data);

The Read will read as many bytes as the length of the byte[].

Dweaver309, 2018-07-18:
Good Morning Adrian, I may have it working. The first issue was the delay between the write and read statements. The second seems really strange but defining a string to send works but sending a quoted string does not. Example: String str = "This is an EEprom test for ESP32.";
           EEprom24LC256.Write(EEprom24LC256.Address.TestWrite, str);

This does not work:  EEprom24LC256.Write(EEprom24LC256.Address.TestWrite, "This is an EEprom test for ESP32.");?????

I am still testing but I think it is working. I will clean up the working code and post. I am not sure the issues above are accurate but with the changes I think it will work. Thanks to everyone who helped!!!

closed 2018-07-18 by josesimoes