nanoframework/Home #365: Visual Studio Extension doesn't work with 2 or more devices

AdrianSoundy, 2018-07-19:
When you connect 2 Esp32 devices to the VS extension you see the devices in the device explorer but if you try to ping each device it gets confused.

Seems to work once then you get:-

```
2018-07-17 22:09:08Z [NanoDevices: Unknown error opening \\?\USB#VID_10C4&PID_EA60#5&2c3064f&0&3#{86e0d1e0-8089-11d0-9ce4-08003e301f73}, possibly opened by another app]
No reply from Silicon Labs CP210x USB to UART Bridge (COM6) @ COM6
```

Didn't try anything else

6 replies

josesimoes, 2018-07-19:
Most likely the device isn't being "closed" when the selected device event occurs.

josesimoes, 2019-05-15:
This is working now. Tested with ESP32 and STM32F769 devices connected simultaneously. 
A bit slow on occasions, tough…

closed 2019-05-15 by josesimoes