nanoframework/Home #367: Add target for NXP FRDM-K64F

FusedShadowX, 2018-07-20:
## Details about Problem

nanoFramework area (Hardware/target board | Community targets )

Hello everyone, I recently bought a FRDM-K64F which is not supported by the framework. I know it is not going to be introduced so, however if some people have the knowledge and want to help add the above target, I would be very pleased to help. I don't have the knowledge to add targets but with some guidance I can help too.

piwi1263, 2018-07-23:
Hi, what RTOS is mostly used or the one you are mostly comfortable with?

FusedShadowX, 2018-07-23:
Hello, thank you for the response.
I'm currently using Mbed OS which supports Keil RTX... However, I'm not common with this or any RTOS, so I think that I should directly begin with ChibiOS.

josesimoes, 2018-10-09:
@FusedShadowX any progresses on this?