nanoframework/Home #370: Add startup guide for STM32 devices requiring DFU

josesimoes, 2018-07-24:
Update the guide [here](http://docs.nanoframework.net/articles/getting-started-guides/getting-started-managed.html) to include a description for devices using DFU.

Markup document [here](https://github.com/nanoframework/nanoframework.github.io/blob/pages-source/articles/getting-started-guides/getting-started-managed.md).

josesimoes, 2018-07-24:
Suggestions:
- Add another section in "The hardware" explaining that this tutorial will address the STM32 based targets with details for using ST-Link and DFU, depending on the target availability.
- Similarly in "Uploading the firmware to the board"

josesimoes, 2018-08-29:
@Dweaver309 added this [guide](https://github.com/Dweaver309/UpdateFirmware/blob/master/UpdateFirmware.pdf) to its repo

guess  that can be a good starting point to write our own. 😉 
Guess a bit of copy/paste won't cause any hard feelings.... 😁

Dweaver309, 2018-08-29:
[How to update firmware.zip](https://github.com/nanoframework/Home/files/2332320/How.to.update.firmware.zip)
This is the original document use it anyway you want.  I hope it helps!

Dweaver309, 2019-01-23:
@Dweaver309 I will update the issue.

Dweaver309, 2019-01-23:
I think I sent the changes for this?
https://github.com/nanoframework/nanoframework.github.io/blob/pages-source/articles/getting-started-guides/getting-started-managed.md#uploading-the-firmware-to-the-board

Dweaver309, 2019-01-23:
[create-dfu-file.zip](https://github.com/nanoframework/Home/files/2788239/create-dfu-file.zip)
If the Pull Request didn't make it here are the changes. Thanks for the help

josesimoes, 2019-01-24:
Completed with nanoframework/nanoframework.github.io#15.

closed 2019-01-24 by josesimoes