nanoframework/Home #383: GPIO NULL Reference

Dweaver309, 2018-08-17:
I wish I could give more information but this is what I have.

I upgraded to the 5.0.24 nanoFramework VS2017 extension yesterday and got null reference errors in my WiFi program https://github.com/Dweaver309/nanoFramework.serial.wifi.esp8266

I tried to recreate today with a small simple app and got sporadic errors.

piwi1263, 2018-08-17:
Let me check mine. BTW, what board are you using ?

piwi1263, 2018-08-17:
Output of mine ... still working on 0.5.0.24 & VS 2017 15.7.6

[image]

piwi1263, 2018-08-17:
This is on a IngenuityMicro Oxygen+Neon 

Oxygen is STM32F411CEU6
Neon is a modded ESP8266-03 with 32 Mbit Flash with a very up-2-date AT version.

Dweaver309, 2018-08-18:
Good morning Peter,

I am using the Netduino3 WiFi. With the same software as I published?

Thanks for the help

Dweaver309, 2018-08-18:
The null reference is caused when the output pins are created to output to the LED's .

josesimoes, 2018-08-18:
Could you post here the code? From the gpio provider to the pin write please.

Dweaver309, 2018-08-18:
[NFApp5.zip](https://github.com/nanoframework/Home/files/2299622/NFApp5.zip)

Dweaver309, 2018-08-18:
I am working to find where  the issue started.  Fails with 5.0.24 5.0.22 5.0.18 Works with 5.0.11. I have not tried the versions between 5.0.11 and 5.0.18. I am doing that now to find where the issue started.

Thanks for the help!

piwi1263, 2018-08-19:
It seems that the GpioController is not active yet, it is null. In my simple construction, single source tester of the boards capabilities, it works since the user LED is blinking. Even the interop to get the MCU serial number goes well. Tested on VS 15.7.6 & Extension 0.5.0.24.

josesimoes, 2018-08-22:
@Dweaver309 here's the reason why you're seeing these `NULL`s: you're trying to get the GpioController _AND_ opening a Gpio pin in a static field of a class.
You have to move those lines opening the Gpio pins into the constructor or into that Initialize method.

As a side comment: nanoFramework Gpio API offers a [Toggle](http://docs.nanoframework.net/api/Windows.Devices.Gpio.GpioPin.html#Windows_Devices_Gpio_GpioPin_Toggle) method, so feel free to use that one and have the platform working for you! :wink

Dweaver309, 2018-08-22:
using System;
using System.Threading;
using Windows.Devices.Gpio;

namespace NFApp8
{
    public class Program
    {
        // User pin number
        const int LedNumber = 10;

        // Go Ports pin numbers
        const int Led1Number = 73;
        const int Led2Number = 75;
        const int Led3Number = 78;

        // Leds to toggle
        private static  GpioPin LEDUser;

        private static  GpioPin LED1;

        private static GpioPin LED2;

        private static  GpioPin LED3;

        public static void Main()
        {
            Console.WriteLine("Hello world!");
            
            LEDUser = GpioController.GetDefault().OpenPin(LedNumber);

            LEDUser.SetDriveMode(GpioPinDriveMode.Output);

            LED1 = GpioController.GetDefault().OpenPin(LedNumber);

            LED1.SetDriveMode(GpioPinDriveMode.Output);

            LED2 = GpioController.GetDefault().OpenPin(LedNumber);

            LED2.SetDriveMode(GpioPinDriveMode.Output);

            LED3 = GpioController.GetDefault().OpenPin(LedNumber);

            LED3.SetDriveMode(GpioPinDriveMode.Output);

            LED1.Toggle();

            LED2.Write(GpioPinValue.High);

            LEDUser.Write(GpioPinValue.High);

            Thread.Sleep(Timeout.Infinite);
        }
    }
}

Dweaver309, 2018-08-22:
The code above does not work. I think there is a bug.

josesimoes, 2018-08-22:
When you write "does not work" can you be more specific on what exactly is not working?

Dweaver309, 2018-08-22:
Jose,
The led's do not light up. After one, two or three tries VS gives error can not copy files.  You may get them to work (I was not able to) The next time you compile you might get a NULL reference error. I think in my testing I have seen it all. I even had to reinstall the firmware a couple of times.

josesimoes, 2018-08-22:
I did a quick check on the ND3 WiFi [schematic](https://github.com/WildernessLabs/Netduino_Hardware/blob/master/N3_WiFi/N3_WIFI_13DEC16_0800.pdf) and it seems that the called LED_GOPORT1, 2 and 3 are connected to PD13, 14 and 15 which correspond to pins 60, 61 and 62. 
The code above seems to be using other pins which can lead to erratic behaviour depending on what they are connected too...

Dweaver309, 2018-08-22:
The schematic you referenced shows:

Goport1Led = PE9 - 73
Goport2Led = PE11 - 75
Goport3Led = PE14 -78

The print is way to small.

piwi1263, 2018-08-22:
He Guys, you'll never get a go port led lit this way you are referencing/assigning always to the onboard LED. All LED open pins go against pinnumber 10. The Led1Number, Led2number, Led3Number are never used, even if they have the correct pinnumber this won't work. And taking multiple GpioPin to the output might even complete not working since the exclusiveness of the pin assignment.

piwi1263, 2018-08-22:
I've modified the above mentioned source a bit and now the LEDs for GoPort1, 2, 3 and the on-board LED are lit up one after each other second by second.

```
using System;
using System.Threading;

using Windows.Devices.Gpio;

namespace Blinky
{
    public class Program
    {
        // User pin number
        const int LedNumber = 10;

        // Go Ports pin numbers
        const int Led1Number = 73;
        const int Led2Number = 75;
        const int Led3Number = 78;

        // Leds to toggle
        private static GpioPin LEDUser;
        private static GpioPin LED1;
        private static GpioPin LED2;
        private static GpioPin LED3;

        public static void Main()
        {
            Console.WriteLine("Hello world!");

            LEDUser = GpioController.GetDefault().OpenPin(LedNumber);
            LEDUser.SetDriveMode(GpioPinDriveMode.Output);

            LED1 = GpioController.GetDefault().OpenPin(Led1Number);
            LED1.SetDriveMode(GpioPinDriveMode.Output);

            LED2 = GpioController.GetDefault().OpenPin(Led2Number);
            LED2.SetDriveMode(GpioPinDriveMode.Output);

            LED3 = GpioController.GetDefault().OpenPin(Led3Number);
            LED3.SetDriveMode(GpioPinDriveMode.Output);

            Thread.Sleep(1000);
            LED1.Toggle();

            Thread.Sleep(1000);
            LED2.Write(GpioPinValue.High);

            Thread.Sleep(1000);
            LED3.Write(GpioPinValue.High);

            Thread.Sleep(1000);
            LEDUser.Write(GpioPinValue.High);

            Thread.Sleep(Timeout.Infinite);
        }
    }
}
```

Using VS2017 15.7.6 with latest packages from myget and extension 0.5.0.24

Dweaver309, 2018-08-23:
Thanks Jose I will put the initialization in the constructor. Peter great job finding the timing issue. I will update my code.

josesimoes, 2018-08-24:
@Dweaver309 have you successfully tested the suggested changes? If you're happy it that please close the issue. 
(this is billed by the hour you know... 🤣 😉 )

MikroBusNet, 2018-08-24:
No need to wait. Here is a sample that reproduces the issue with latest build on Quail (preview 776 and extension 0.5.0.24) :

```
using System;
using System.Threading;
using Windows.Devices.Gpio;

namespace gpioController_null
{
    public class Program
    {
        public static void Main()
        {
            // Non-static class test
            //new Test().Led2.Write(GpioPinValue.High);

            // Static class test
            TestStatic.Led2.Write(GpioPinValue.High);

            Thread.Sleep(Timeout.Infinite);
        }
    }

    /*
    public class Test
    {
        public readonly GpioController gpioController;
        public readonly GpioPin Led1;
        public readonly GpioPin Led2;
        public readonly GpioPin Led3;

        public Test()
        {
            Console.WriteLine("Non-static class init started");
            gpioController = GpioController.GetDefault();

            Led1 = gpioController.OpenPin(4*16+15);         // PE15
            Led1.SetDriveMode(GpioPinDriveMode.Output);
            Led1.Write(GpioPinValue.High);
            Led2 = gpioController.OpenPin(4*16+10);         // PE10
            Led2.SetDriveMode(GpioPinDriveMode.Output);
            Led3 = gpioController.OpenPin(2*16+3);          // PC3
            Led3.SetDriveMode(GpioPinDriveMode.Output);
        }
    }
    */
    
    public static class TestStatic
    {
        public static readonly GpioController gpioController;
        public static readonly GpioPin Led1;
        public static readonly GpioPin Led2;
        public static readonly GpioPin Led3;

        static TestStatic()
        {
            Console.WriteLine("Static class init started");
            gpioController = GpioController.GetDefault();

            Led1 = gpioController.OpenPin(4 * 16 + 15);         // PE15
            Led1.SetDriveMode(GpioPinDriveMode.Output);
            Led2 = gpioController.OpenPin(4 * 16 + 10);         // PE10
            Led2.SetDriveMode(GpioPinDriveMode.Output);
            Led3 = gpioController.OpenPin(2 * 16 + 3);          // PC3
            Led3.SetDriveMode(GpioPinDriveMode.Output);
            Led3.Write(GpioPinValue.High);
        }
    }
    
}
```

You will get an exception on the line `Led1 = gpioController.OpenPin(4 * 16 + 15);         // PE15` in the static class.
If you uncomment the non-static class parts and comment the static ones, then everything is fine.
This was working "before".

Dweaver309, 2018-08-24:
Christophe thanks for the help. I have updated my code.

MikroBusNet, 2018-08-24:
Not sure you should close this issue...

Dweaver309, 2018-08-24:
I think you are right I tried using your code in an other class object and
got a null reference. I just rewrote the code and I am having trouble
making pin 10 go low??? However I think someone else should press the issue
that is more knowledgeable then me. I have seen your code you have a much
better understanding of what is happening than I do. Thanks again!

MikroBusNet, 2018-08-24:
Thank you for your recommendations.

MikroBusNet, 2018-08-24:
Last chance :

```
namespace gpioController_null
{
    public class Program
    {
        public static void Main()
        {
            TestStatic.Led1.Write(GpioPinValue.High);

            Thread.Sleep(Timeout.Infinite);
        }
    }

    public static class TestStatic
    {
        public static GpioPin Led1;

        static TestStatic()
        {
            Led1 = GpioController.GetDefault().OpenPin(4 * 16 + 15);         // PE15
            Led1.SetDriveMode(GpioPinDriveMode.Output);
        }
    }
}
```

Dweaver309, 2018-08-25:
Jose,
GPIO is working with my WiFi driver. The issues I explained above were do to trying to init. GPIO ports after the constructor setup the class object. It appears as you said the init. must be done in the constructor. The code has been updated here: https://github.com/Dweaver309/nanoFramework.serial.wifi.esp8266

Thanks for the help!

MikroBusNet, 2018-08-25:
José, just to be sure and clear...

GPIO, I2C and SPI controllers (those I have tested) can't be used in static constructors. Even a call like `var Led1 = GpioController.GetDefault().OpenPin(4 * 16 + 15);` in the static constructor does produce a NULL exception.
So this mean that gpio, i2c or spi (at least) can't be used in nanoFramework in static classes or as static fields of a class (static or not).
Which in turn means that you have to create a non-static class and instantiate it to be able use its fields (static or not).

But... This also means that you can't benefit of a static constructor in such class.

Consider the following code :
```
using System;
using System.Threading;
using Windows.Devices.Gpio;
using Windows.Devices.Spi;

namespace gpioController_null
{
    public class Program
    {
        public static void Main()
        {
            var c1 = new MyClass(11);

            Console.WriteLine("myInt1 = " + MyClass.myInt1);  // Static field
            Console.WriteLine("c1.myInt2 = " + c1.myInt2);  // Instance field
            
            //MyClass.Led1.Write(GpioPinValue.High);
            MyClass.Led2.Write(GpioPinValue.High);

            Thread.Sleep(Timeout.Infinite);
        }
    }
    
    public class MyClass
    {
        public static GpioPin Led1;
        public static GpioPin Led2;
        public static Int32 myInt1;
        public Int32 myInt2;

        static MyClass()
        {
            myInt1 = 999;
            Led1 = GpioController.GetDefault().OpenPin(4 * 16 + 15);         // PE15
            Led1.SetDriveMode(GpioPinDriveMode.Output);
            Led1.Write(GpioPinValue.High);
        }

        public MyClass(int number)
        {
            myInt2 = myInt1-number;

            Led2 = GpioController.GetDefault().OpenPin(4 * 16 + 10);         // PE10
            Led2.SetDriveMode(GpioPinDriveMode.Output);
        }
    }
}
```
Led1 will always be null because of the exception in the static constructor. So I can't use it.
Now, if I comment the Led1 code in the static constructor, I get no exception but Led2 won't be available until I instanciate the class. In this case, the class constructor will initialize Led2 without exception.

So, unless José says that static "things" are hell and that they should not be used with nanoFramework, then there is a bug somewhere.

And, everyone seems to forget that, this was working until at least preview 696, the last one I have used before testing 766. The MBN "core" has always been working with nf until now and it has always used static "gpio" fields (onboard leds, mainly).
This is also working fine with TinyCLR.

Dweaver309, 2018-08-26:
Christophe I think you make some valid points and this issue should be considered again. I am reopening it. I  hope the code will be rolled back to were it was before?

josesimoes, 2018-08-27:
Found the root cause for this. I'll try to explain without going into too much technical details.

When the CLR starts it starts by searching for the aseembly that holds the application executable (that's the `Program.Main()`).
The next step is to look for the assemblyes that that assembly references and queues them for execution.
The next step is to look for static constructors on that assembly and executes them.

In the situation above, the static constructor for `MyClass()` is executed _before_ `Main()` because  `MyClass()` is used by `Main()` therefore it needs to be constructed before being used.

When the static constructor of `MyClass()` starts executing the classes (and static constructor) of the Windows.Devices.Gpio assembly are queued for execution _but_ haven't been called yet so the s_instance of `GpioController` is null and leads to the situation reported.

If you move `MyClass` to a separate project and reference it from the executable it works as expected. Why? Because the `Program.Main()` assembly is referencing `MyClass` assembly (which will be executed _before_ `Main` and `MyClass` is referencing Windows.Devices.Gpio assembly, which will be executed _before_ the static constructor of `MyClass`. When the execution reaches the static construtor of `MyClass` the constructor of `GpioController` has already executed and the s_instance there will be properly instanciated and the execution will run smoothly.

This workflow hasn't been changed from the beginig so I'm guessing that the fact that MBN core was working "before" (as reported) is due to the fact that it lives on it's own assembly and executes in the proper order as explained above. So it should be working fine now as it was before. At least as far as this situation is concerned. It's not to exclude that something else has changed on the assemblies it depends on and causing other issues.

closed 2018-08-27 by josesimoes