nanoframework/Home #384: WatchChar is not implemented in ESP32

Dweaver309, 2018-08-22:
The WatchChar is not implemented in Windows.Devices.SerialCommunication for ESP32.

MatthiasJentsch, 2018-08-26:
I can't find out how the event gets from PostManagedEvent routed to the correct event handler in C#.

MatthiasJentsch, 2018-08-26:
The last code is here, but it's not working I call PostManagedEvent and the C# event handler will not b3e called.
https://github.com/MatthiasJentsch/nf-interpreter/blob/watchchar/targets/FreeRTOS/ESP32_DevKitC/nanoCLR/Windows.Devices.SerialCommunication/win_dev_serial_native_Windows_Devices_SerialCommunication_SerialDevice.cpp

josesimoes, 2018-08-27:
I think the code is missing is the native interrupt wrappers. Wrote this as a reminder. See [here](https://github.com/nanoframework/Home/blob/master/docs/architecture/native-interrupt-handlers.md).
As it seems that that is not a _true_ interrupt handler, pherhaps the closest would be to wrap the switch statment there (line 107) and 125-126.

MatthiasJentsch, 2018-08-28:
Thanks @josesimoes for the hint with the portIndex. For UART0 the portIndex 1 should be send because COM1 is UART0 in the ESP32.

closed 2018-08-28 by MatthiasJentsch